jrsteensen/OpenHornet #21: Parts List

jrsteensen, 2019-10-16:
Create a parts list for the major end items and top level assembly.

- [ ] Left Console
- [ ] Lower Instrument Panel
- [ ] Upper Instrument Panel
- [ ] Right Console
- [ ] Center Tub
- [ ] SJU-17 Ejection Seat
- [ ] Top Level Assembly

ScottyMac52, 2019-10-20:
This may not be the right place to ask this but I am looking to build just a simple F/A-18C instrument panel, UFC etc. out of MDF but I am having an extremely hard time sourcing blueprints or drawings?

jrsteensen, 2019-10-20:
> This may not be the right place to ask this but I am looking to build just a simple F/A-18C instrument panel, UFC etc. out of MDF but I am having an extremely hard time sourcing blueprints or drawings?

Hop on our discord - that's a more appropriate place to discuss. Right now this project doesnt have any drawings available.

jrsteensen, 2020-04-27:
Moving this to Top Level Release Package.

closed 2020-04-27 by jrsteensen