jrsteensen/OpenHornet #36: UFC PCB

jrsteensen, 2019-10-20:
- [ ] Requirements definition
- [ ] Schematic completed
- [ ] PCB routed
- [ ] Schematic/PCB reviewed
- [ ] BOM generated
- [ ] Gerbers generated
- [ ] MCAD generated

amanuense, 2019-10-23:
UFC revB is complete

jrsteensen, 2019-10-29:
UFC requires full manufacturing file creation.

jrsteensen, 2019-11-05:
PR #73 addresses this. Closed.

jrsteensen, 2019-11-24:
Whelp, MAX6953 IC is obsolete and end of life. Going to have to redesign - either find suitable new ICs to control option displays, or source new displays. Either way a major redesign is required.

amanuense, 2019-11-24:
That's a bummer. Next week we launch my current project. I'm going to get
some free time once I finish documentation, trainings, and some plans. I'll
be able to make time to check that then.
It will be a good opportunity to work on my idea of the mezzanine and make
it modular

jrsteensen, 2019-11-25:
Agreed to a degree! I've also working on the MCAD to allow for discrete LEDs so we can replicate the real displays a little more closely. Only trade off is we will likely have to have the LEDs done when fabbed via jlc or similar.

[image]

[image]

jrsteensen, 2019-12-06:
Starting UFC 2.0. Will open new issue with new requirements.

closed 2019-12-06 by jrsteensen