jrsteensen/OpenHornet #41: DDI/AMPCD Arduino Shield

jrsteensen, 2019-10-20:
- [ ] Requirements definition
- [ ] Schematic completed
- [ ] PCB routed
- [ ] Schematic/PCB reviewed
- [ ] BOM generated
- [ ] Gerbers generated
- [ ] MCAD generated

amanuense, 2019-10-23:
found we are missing a pullup for the IRQ line. we are also missing bypass caps for the ICs

amanuense, 2019-10-23:
missing rotaty switch, created issue #52 to track this separately

jrsteensen, 2019-10-23:
> found we are missing a pullup for the IRQ line. we are also missing bypass caps for the ICs

This references the DDI and AMPCD bezel (#32, #33) not this issue.

jrsteensen, 2019-10-23:
> missing rotary switch, created issue #52 to track this separately

TODO for this PCB:

- [ ] Remove 1x POT connector
- [ ] Add 4 pin connector for DDI rotary (3 DI + GND)
- [ ] Add 2x 3 pin connector (2 DDI + GND) for AMPCD HDG/CRS toggles
- [ ] Fix i2c addressing.

jrsteensen, 2019-10-24:
Bug fixed with PRs #54 and #55. Ready for review, Gerber generation, and MCAD generation.

jrsteensen, 2019-10-29:
Completed in PR #64.

closed 2019-10-29 by jrsteensen