jrsteensen/OpenHornet #45: Power Distribution Module

jrsteensen, 2019-10-20:
- [ ] Requirements definition
- [ ] Schematic completed
- [ ] PCB(s?) routed
- [ ] Schematic/PCB reviewed
- [ ] BOM generated
- [ ] Gerbers generated
- [ ] MCAD generated

jrsteensen, 2019-10-21:
## Requirements: (WIP)
- [ ] 2X ATX Inputs (Just in case)
- [ ] Thick Traces
- [ ] Thick Copper
- [ ] Terminal Board for displays, and misc components.
- [ ] Connectors for each downstream power bus.

amanuense, 2019-10-23:
do we have size requirements?

jrsteensen, 2019-10-24:
> do we have size requirements?

- [ ] Enclosure Size: 12L X 18W X 5.75H

jrsteensen, 2019-10-24:
## UIP Accessories
Component | Type/PN | PDU Connector Ref Desg | Power Bus ID | Cable Ref Desg | Component Connector Type | 3.3VDC Peak (mA) | 3.3VDC Typ (mA) | 5VDC Peak (mA) | 5VDC Typ (mA) | 12VDC Peak (mA) | 12VDC Typ (mA) | Remote Power Switch | Notes
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
L DDI Display | ER-TFTV101A1-1 | J1 |   |   | Barrel, Center Hot |   |   | 1000 |   |   |   | INT DISP |  
R DDI Display | ER-TFTV101A1-1 | J1 |   |   | Barrel, Center Hot |   |   | 1000 |   |   |   | INT DISP |  
AMPCD Display | ER-TFTV101A1-1 | J1 |   |   | Barrel, Center Hot |   |   | 1000 |   |   |   | INT DISP |  
Sabrent 13 Port USB2.0 Hub | HB-U14P | J1 |   |   | Barrel, Center Hot |   |   | 3000 |   |   |   | INT DISP |  
Standby/SARI/RWR Display | ER-TFTV101A1-1 | J2 |   |   | Barrel, Center Hot |   |   | 1000 |   |   |   | INT DISP |  
IFEI Display | ER-TFTV070A1-6 | J2 |   |   | Barrel, Center Hot |   |   | 1000 |   |   |   | INT DISP |  
RPi 1: L DDI, R DDI | RPI4-MODBP-2GB | J3 |   |   | USB-C |   |   | 3000 |   |   |   | INT DISP |  
RPi 2: AMPCD, Standby/SARI/RWR | RPI4-MODBP-2GB | J3 |   |   | USB-C |   |   | 3000 |   |   |   | INT DISP |  
Rpi 3: IFEI | RPI4-MODBP-2GB | J3 |   |   | USB-C |   |   | 3000 |   |   |   | INT DISP |  
Netgear 5 Port ETH Switch | GS105v5 | J4 |   |   | Barrel, Center Hot |   |   |   |   | 500 |   | INT DISP |

amanuense, 2019-10-24:
12mm x 18mm x 7.75mm is too small. is that a power module for ants?
Oscar

jrsteensen, 2019-10-25:
Inches! Its a nuclear facility for ants!

amanuense, 2019-10-25:
I assume you never watched Zoolander

jrsteensen, 2019-10-25:
[PDUPowerRequirements.xlsx](https://github.com/jrsteensen/OpenHornet/files/3772918/PDUPowerRequirements.xlsx)

L-Walker, 2020-10-27:
[image][image]

jrsteensen, 2022-07-21:
Interesting links:
https://www.amazon.com/Screw-Mount-Position-Distribution-Module/dp/B08LPS7HPX
https://www.sainsmart.com/products/16-channel-12v-relay-module

Add volt/ammeter to each bus