jrsteensen/OpenHornet #108: EVU-F2MFL3B14 pot obsolescence issue

jrsteensen, 2019-11-24:
EVU-F2MFL3B14 POT is obsolete. Need to locate suitable replacement and replace throughout the sim.

jrsteensen, 2019-11-24:
Will replace with:

P0915N-FC20BR10K

https://www.digikey.com/product-detail/en/tt-electronics-bi/P0915N-FC20BR10K/987-1762-ND/6155634

balzreber, 2019-12-05:
The P0915N-FC20BR10K don't seem to have the same screw (shaft) diameter than the old ones. At least they don't fit into the OH1A8A2A-11 - HUD Panel Backplate. (They have a considerably smaller diameter)

balzreber, 2019-12-08:
I'm referring to the bushing size. But it is almost the same as the rest of the shaft size in this model.

balzreber, 2019-12-08:
Also, there is a small key thingy on top of the housing.
[image]

jrsteensen, 2019-12-16:
Added CAD model of new pot with commit 92bd91f.

jrsteensen, 2019-12-16:
Closing issue. Will be addressed in each Release Package issue.

closed 2019-12-16 by jrsteensen