jrsteensen/OpenHornet #117: UFC 2.0 PCB

jrsteensen, 2019-12-06:
- [ ] Requirements definition
- [ ] Schematic completed
- [ ] PCB routed
- [ ] Schematic/PCB reviewed
- [ ] BOM generated
- [ ] Gerbers generated
- [ ] MCAD generated

jrsteensen, 2019-12-06:
# UFC 2.0 Requirements

1. MEGA Shield (SMD headers on back)
1. OH Standard PWR Header (4 Pin Polarized SMD Header)
1. INPUTS:
    1. Snap Dome Switches (GX06400)
        1. 1, 2, 3, OPT 2
		1. 4, 5, 6, OPT 3
		1. 7, 8, 9, OPT 4
		1. CLR, 0, ENT, OPT 5
		1. OPT 1
		1. I/P
		1. EMCON
    1. Encoders with Pushbutton (PEC11R-4220F-S0012)
        1. COMM 1
        1. COMM 2		
    1. Potentiometers (P0915N-FC20BR10K)
	    1. BRT
		1. COMM 1
		1. COMM 2
	1. TOGGLE, 3 POS (100SP3T2B3M2RE)
	    1. ADF
1. OUTPUTS
    1. Scratchpad (ER-OLEDM023-1W OLED Display)
	1. OPT Line 1
		1. 4x DIY 16 seg displays (64x 16-219A/T2D-AR2T1QY/3T)
		2. Option Indicator (2x 16-219A/T2D-AR2T1QY/3T) - controlled by digit 1 & 2 DP on MAX6955?)
	1. OPT Line 2
		1. 4x DIY 16 seg displays (64x 16-219A/T2D-AR2T1QY/3T)
		2. Option Indicator (2x 16-219A/T2D-AR2T1QY/3T) - controlled by digit 1 & 2 DP on MAX6955?)
	1. OPT Line 3
		1. 4x DIY 16 seg displays (64x 16-219A/T2D-AR2T1QY/3T)
		2. Option Indicator (2x 16-219A/T2D-AR2T1QY/3T) - controlled by digit 1 & 2 DP on MAX6955?)
	1. OPT Line 4
		1. 4x DIY 16 seg displays (64x 16-219A/T2D-AR2T1QY/3T)
		2. Option Indicator (2x 16-219A/T2D-AR2T1QY/3T) - controlled by digit 1 & 2 DP on MAX6955?)
	1. OPT Line 5
		1. 4x DIY 16 seg displays (64x 16-219A/T2D-AR2T1QY/3T)
		2. Option Indicator (2x 16-219A/T2D-AR2T1QY/3T) - controlled by digit 1 & 2 DP on MAX6955?)
	1. COMM 1 Display
        1. 1x DIY 16 seg displays (16x 16-219A/T2D-AR2T1QY/3T)
	1. COMM 2 Display
        1. 1x DIY 16 seg displays (16x 16-219A/T2D-AR2T1QY/3T)
	1. 16 Segment Display Drivers (4X MAX6955 i2c Driver)
		1. IC1 controls OPT 1&2
		1. IC2 controls OPT 3&4
		1. IC3 controls OPT 5 & COMM 2
		1. IC4 controls COMM 1
1. EXTERNAL COMPONENTS
May combine into single header if back face space allows.
    1. HUD Panel (2x8 Header)
	1. AP Panel (2x4 Header)
	1. HUD BIT Panel (1x4 Header)

amanuense, 2019-12-06:
>    2. OH Standard PWR Header (4 Pin Polarized SMD Header)
>    3. INPUTS:
>          2. 4, 5, 6, OPT 3
>          3. 7, 8, 9, OPT 4
>          4. CLR, 0, ENT, OPT 5
>          5. OPT 1
>          7. EMCON
>       2. Encoders with Pushbutton (PEC11R-4220F-S0012)
>          2. COMM 2
>       3. Potentiometers (P0915N-FC20BR10K)
>          2. COMM 1
>          3. COMM 2
>       4. TOGGLE, 3 POS (100SP3T2B3M2RE)
>       4. OUTPUTS
>       2. OPT Line 1
>       3. OPT Line 2
>       4. OPT Line 3
>       5. OPT Line 4
>       6. OPT Line 5
>       7. COMM 1 Display
>       8. COMM 2 Display
>       9. 16 Segment Display Drivers (4X MAX6955 i2c Driver)
>          2. IC2 controls OPT 3&4
>          3. IC3 controls OPT 5 & COMM 2
>          4. IC4 controls COMM 1
>       5. EXTERNAL COMPONENTS
>       2. AP Panel (2x4 Header)
>       3. HUD BIT Panel (1x4 Header)

I will soon have time to work on the project.
 Create a item for a mezzanine board that will help a lot with routing.

jrsteensen, 2019-12-06:
Updated. I am starting work on the schematic, I'll let you know when I need inputs/review.

amanuense, 2019-12-06:
Cool I'll take a look to good board connectors.

The idea of the mezzanine is to use it to simplify routing or to have
bigger area for connectors.

I want to do both

jrsteensen, 2019-12-07:
[image]

Can I get a quick review on this against the MAX6955 datasheet? DP LED will not be populated in any of the displays, but I think leaving it NC screws up the charlieplexing.

https://datasheets.maximintegrated.com/en/ds/MAX6955.pdf

jrsteensen, 2019-12-08:
@amanuense Schematic is ready for review.

https://www.maximintegrated.com/en/products/power/display-power-control/MAX6955.html

amanuense, 2019-12-09:
review notes:
key matrix GPIO assignation was missing
no 5v to 3.3v I2C translation
LED controller oscillators not synced.

I took the liberty of making these changes, you can observe the diff here
https://www.dropbox.com/s/cld2v8p44ih4f7m/diff_UFC_3.039802a.pdf?dl=0

I will push the changes tomorrow if there is no more updates.

jrsteensen, 2019-12-14:
See attached ZIP for drawing and DXF for PCB.

[UFC PCB REQUIREMENTS.zip](https://github.com/jrsteensen/OpenHornet/files/3964283/UFC.PCB.REQUIREMENTS.zip)

Please let me know when components are placed - will review before routing.

jrsteensen, 2019-12-18:
Please see attached CSV file with XYZ coordinates of the LEDs.

[UFC LED SKETCH.zip](https://github.com/jrsteensen/OpenHornet/files/3980901/UFC.LED.SKETCH.zip)

amanuense, 2019-12-19:
X, Y, and Z... Fancy

amanuense, 2020-01-20:
restarting efforts on this issue.

amanuense, 2020-01-20:
made correction to the UFC LED sketch. the document had too many LEDs (435) vs the projected 362 LEDs (including opt), I have attached a spreadsheet where I matched the LED reference on Eeschema, location, absolute, and relative position.

the spreadsheet contains a graph for the x,y location of each LED on the first page.

this change is accompanied with changes to the schematic, this will help me automate the positioning of the LEDs in the PCB.
[UFC LED SKETCH.xlsx](https://github.com/jrsteensen/OpenHornet/files/4085676/UFC.LED.SKETCH.xlsx)

jrsteensen, 2020-05-06:
@amanuense Do you have a status update on this?

amanuense, 2020-05-07:
I have finished probably 50% of the routing. I have been busy at work but I
am making slow progress.
Oscar

jrsteensen, 2020-11-10:
This will feed into the ABSIS 2.0 based UFC, and a lot of the hard work is complete here. Do not despair @amanuense. All is not for naught.

closed 2021-01-05 by jrsteensen