jrsteensen/OpenHornet #119: Check ABSIS functionality

balzreber, 2019-12-09:
Full functionality check of both the ABSIS MEGA and ABSIS NANO boards.

petersawka, 2021-02-13:
ABSIS Nano Board Tested for USB Communication to PC for code Loading:
[image]

Encountered an issue with USB Cable and BUS-OUT Molex Connector:
[image]

Additional tests are required for Nano Programming when not installed at EOL.

jrsteensen, 2021-02-13:
So, the USB connector vs MOLEX connector interference would be easily fixed using a slimline usb connector or a 90* adapter.

The simplest solution, and I would like to get input here to solve both issues is rather than use JP1, go ahead and disconnect the nano/mega from the ABSIS PCB for programming.

What are your thoughts? @petersawka @L-Walker @amanuense

petersawka, 2021-02-13:
@jrsteensen I personally am all for removing the Nano for programming; I think this solves both issues, and in some locations in the pit it would more beneficial.

jrsteensen, 2021-07-27:
To summarize: JP1 does not function as expected. Current workaround will be to remove the Nano from the PCB for programming until PCB is revised at some point in the future. Bumping this to a beta3 issue, since it is not a limiting factor.

closed 2021-07-27 by jrsteensen