jrsteensen/OpenHornet #123: ABSIS MEGA MASTER

balzreber, 2019-12-11:
We need an ABSIS MEGA MASTER as the last board in the RS485 chain. The current ABSIS MEGA uses Serial0 for RS485 communication but also uses the same Serial0 for USB communication with DCS-bios. The ABSIS MEGA MASTER should use Serial0 for USB and Serial1, 2 or 3 for RS485.

jrsteensen, 2019-12-11:
I believe this should be done on the existing MEGA board so we don't add yet another type of PCB to the repo. For the Master, we just don't fully populate the ABSIS Mega, instead just the RS485 critical components.

balzreber, 2019-12-11:
I don't know if that is possible since we have to reroute Serial1, 2 or 3 to the RS485 components.

jrsteensen, 2019-12-11:
> I don't know if that is possible since we have to reroute Serial1, 2 or 3 to the RS485 components.

Why? All the megas can use Serial1, 2 or 3 without any significant degradation of performance?

balzreber, 2019-12-11:
I can't find any documentation if it is possible for the slave ones to use Serial1, 2, or 3. Or if they automatically use Serial0.

jrsteensen, 2019-12-11:
If you check the template master sketch in the DCS-BIOS repo, you specify the pins for TX/RX in the sketch, so it shouldn't matter.

balzreber, 2019-12-11:
But that is just the TX_ENABLE pin. Not the Serial. Breadboarding is a great idea. I will do that if I find some time.

balzreber, 2019-12-14:
I can confirm that RS485 only works on Serial1, and not on Serial0. Serial2 and 3 should also work, but I haven't tested them. My test setup was one ABSIS MEGA with the Serial1 pins of the Arduino rerouted to the RS485 pins. And Pin 2 as TX_ENABLE. On the other side was a standard ABSIS NANO. Both had 120Ohm resistors on the RS485 bus.

jrsteensen, 2019-12-14:
So this should be resolved in #125 ?

As a slave, the DIP switches disable the Serial0 USB for using Serial instead for RS485.

balzreber, 2019-12-14:
I haven't checked the MOSFETs yet. But there shouldn't be any Serial changes needed for all ABSIS boards except the master one.

amanuense, 2019-12-14:
Not completely sure. Will discuss tomorrow over discord to understand but
better

balzreber, 2019-12-17:
I tested with two ABSIS MEGA's today. Master can have RS485 on any serial except Serial0. And the slave must have RS485 on Serial0.

amanuense, 2019-12-17:
Is it hw limited or software limited?

jrsteensen, 2019-12-17:
Lets see if we can't have DCS-BIOS implement a configuration variable to allow to use anything other than Serial0 on the Mega for RS485 communcation.

jrsteensen, 2020-06-07:
This was from [FSF}_Ian:

> I think we can add a #define that lets you run the Mega 2560 as an RS-485 slave over Serial1. After a quick look at the code, I think the only part that would need to be changed is the following lines at the end of DcsBiosNgRS485Slave.cpp.inc:
```
    RS485Slave rs485slave(&UDR0, &UCSR0A, &UCSR0B, &UCSR0C, TXENABLE_PIN);    
    ISR(USART0_RX_vect) { rs485slave.rxISR(); }
    ISR(USART0_TX_vect) { rs485slave.txcISR(); }
    ISR(USART0_UDRE_vect) { rs485slave.udreISR(); }
```

@balzreber