jrsteensen/OpenHornet #133: UIP Release Package

jrsteensen, 2019-12-16:
Update all knobs (shaft cavity and set screws).
Replace all EVU-F2MFL3B14 pots with P0915N-FC20BR10K pot.
Ensure all CNC'd parts have internal radii at corners.
Complete UFC MCAD.
Update interconnect diagram.
Complete drawing.
Generate all intermediate/manufacturing files.

jrsteensen, 2019-12-19:
fe689c469882230ef77310b0683da36bab09177e
7b1a562724e2b9bba09d5e14d232d1bc7f32341d
1262a5e80fb313d61c9bef36877d5e6475fc3a96

jrsteensen, 2019-12-24:
f116ba5 - Completed draft of top level drawing. Requires review before proceeding.

[UIP Drawing](https://github.com/jrsteensen/OpenHornet/blob/MCAD_UIP_Release_Package/release/2_Upper%20Instrument%20Panel/OH1A%20-%20ASSY%2C%20UPPER%20INSTRUMENT%20PANEL.PDF)

ajmilner85, 2019-12-24:
Is the asymmetric installation of the threaded inserts, item 64, into UIP LEFT OUTBOARD LONGERON (Sheet 4/C8) and UIP RIGHT OUTBOARD LONGERON (Sheet 4/B5) intentional or was one omitted by mistake? 

I also can't find where the remaining 6 out of the total 14 8-32 threaded inserts are consumed during the assembly. Are there meant to be 3 inserted into the front and back of each longeron?

PR #149 (2019-12-26): MCAD: UIP Release Package
# Description

Major updates to UIP to prep for release package. All drawings complete. Manufacturing files are required.

Closes #133 

## Type of change

- [X] New MCAD (new Mechanical Model)
- [X] Change MCAD (fixed issue with existing MCAD model)

# Checklist: (Delete non-relevant sections)

## MCAD:
- [X] Model is in SolidWorks 2019 format
- [X] Design intent and requirements have been met
- [X] Design has been integrated into Next Higher Assembly (NHA)
closed 2019-12-26 by jrsteensen