jrsteensen/OpenHornet #136: Right Console Release Package

jrsteensen, 2019-12-16:
- [ ] Update all knobs (shaft cavity and set screws).
- [ ] Replace all EVU-F2MFL3B14 pots with P0915N-FC20BR10K pot.
- [ ] Ensure all CNC'd parts have internal radii at corners.
- [ ] Complete all outstanding items assigned in end item project.
- [ ] Update interconnect diagram.
- [ ] Complete drawing.
- [ ] Check for duplicate hardware.
- [ ] Generate all intermediate/manufacturing files.

dnamaster2000, 2020-05-03:
STL files for Right Console Structure, all test printed, high poly, unmodified.

[Right Console Structure.zip](https://github.com/jrsteensen/OpenHornet/files/4572252/Right.Console.Structure.zip)

jrsteensen, 2020-06-30:
Listed all drawings for right console in issue.

jrsteensen, 2021-07-23:
Combined into #313

closed 2021-07-23 by jrsteensen