jrsteensen/OpenHornet #148: UIP Manufacturing Files

jrsteensen, 2019-12-26:
- [ ] Generate all intermediate/manufacturing files. (Will split this off to separate issue.)
    - [ ] All UIP DXFs
    - [ ] All UIP STLs
    - [ ] OH1A-1_SH[SHEET #]_[SHEET THICKNESS]_MDF.DXF (MDF CNC PARTS)
    - [ ] OH1A-LEGENDS.DXF (LEGEND PLATES)
    - [ ] OH1A-LIGHTPLATES.DXF (LIGHT PLATES)
    - [ ] OH1A-BACKPLATES.DXF (BACKPLATES)
    - [ ] OH1A-INDICATORCOVERS.DXF (INDICATOR COVERS)
    - [ ] OH1A-INDICATORLEGENDS.DXF (INDICATOR LEGENDS)
    - [ ] OH1A-DIFFUSERSGREEN.DXF (INDICATOR DIFFUSERS, GREEN)
    - [ ] OH1A-DIFFUSERSAMBER.DXF (INDICATOR DIFFUSERS, AMBER)
    - [ ] OH1A-DIFFUSERSRED.DXF (INDICATOR DIFFUSERS, RED)
    - [ ] All DDI Knobs (Revised for shaft and setscrews)
    - [ ] Spin Panel HMD Knobs (Revised for shaft and setscrews)
    - [ ] All HUD Panel Knobs (Revised for shaft and setscrews)
    - [ ] HUD Panel backplate (Revised for new potentiometer)
    - [ ] HUD BIT Box (Revised to remove UFC Controller PCB and associated HW)
    - [ ] All UFC Panel Primary Module Parts (Integrated UFCv3)

jrsteensen, 2019-12-26:
Please make any commits to the MCAD_UIP_Release_Package branch.

Luft101, 2020-03-01:
the files OH1A-110.stl, OH1A-111.stl and OH1A-116.stl are low-poly, also in the latter, the text is inverted. If you fix the originals, I convert them back to stl when you want

jrsteensen, 2021-07-27:
Merging with #314

closed 2021-07-27 by jrsteensen