jrsteensen/OpenHornet #161: Indicator Legends too small

balzreber, 2020-01-03:
The indicator legends on the Eyebrow Warning Indicators left and right are too small. They should be as big as the glass plates they are mounted on, not as big as the hole in the indicator panel.

[image]

jrsteensen, 2020-01-06:
1152ca970cb183f3a53e607d31d2bd91b6dbb2b1: fixed font sizing in EWI legends. Size should be good, if centered properly. Check size during reprint, perhaps oversize very slightly, then cut to fit. This might be an iterative process. For now, going to close this. Can reopen if not resolved.

balzreber, 2020-01-10:
Reexport all the UIP ScreenPrints after @jrsteensen fixed them.

jrsteensen, 2020-01-13:
Updated .55 indicator in 4a235e0a0d6938a73d6f3fc2111e025e38ab16a8.
Updated spin indicator font in 5ca04192a69a265bf7aa0d3d2c93ef0bf82a8e42

jrsteensen, 2020-01-17:
@balzreber Is this resolved with those two commits?

jrsteensen, 2020-04-16:
@balzreber Do you have any updates on this issue?

closed 2020-04-17 by jrsteensen