jrsteensen/OpenHornet #164: Redesign DDI Shield

balzreber, 2020-01-05:
Redesign DDI Shield to accommodate two rotary encoders instead of potentiometers.

jrsteensen, 2020-01-06:
Done. Requires manufacturing files. Thanks @balzreber and @amanuense

balzreber, 2020-01-10:
Found that the I2C pins are routed to the wrong Arduino Pins. They are D4 and D5 currently and should be changed to A4 and A5. I will have a stab at this.

amanuense, 2020-01-10:
That means the template is wrong we should check other custom pcbs using
i2c including UFC

jrsteensen, 2020-01-17:
Any update on this?

jrsteensen, 2020-05-06:
@balzreber Can you provide a status update on this?

balzreber, 2020-05-22:
I didn't do any work on that. But since I will receive some demo dome switches soon, I hope to assemble the DDI soon and then come back to that error.

balzreber, 2020-06-05:
I have redesigned the shield. But since I tore up some stuff it needs a thorough review before I'm comfortable ordering an alpha version for testing.

jrsteensen, 2020-06-05:
Awesome! Please make a pull request so we can review it. 😁

jrsteensen, 2021-01-05:
This is outdated with the ABSIS 2.0 redesign. Closing this issue.

closed 2021-01-05 by jrsteensen