jrsteensen/OpenHornet #166: Normal Time push button on Sim Control Panel

bjanders, 2020-01-16:
The Sim Control Panel has a toggle switch for increasing and decreasing game time speed. It would be nice with a "Normal Time" push button on the Sim Control panel. When you fast forward with the toggle switch and then reach the point where you want to resume normal time game play I'd rather press a button than slowing it down with the toggle switch until normal speed is reached.

jrsteensen, 2020-01-16:
Will look at redesign, including moving the Master Sim Mode switch to the SIM PWR panel on the left console. Alternatively, could potentially combine by converting some momentary toggles to 4-way hats.

jrsteensen, 2020-01-17:
[image]
WIP

commit e943b70b (2020-01-17): Updated R Console Panels

Updated locking toggles to correct bats throughout right console. #136
Removed master mode from SIM CNTL panel. (Now located on SIM PWR)
Changed SLOW legend to REAL for TIME position on SIM CNTL panel. Closes #166
closed 2020-01-17 by jrsteensen