jrsteensen/OpenHornet #181: MDF thickness

nic0laz, 2020-03-31:
Hi, all the mdf with are in inches and in argentina (and several other countries) all the wood planks come in metric withs, that is 6mm, 9,, 12mm, 16mm, 18mm.

Is it really hard to adapt all the plans (or at least the ejection seat) to use metric mdf?

thanks!

balzreber, 2020-04-05:
Hi nic0laz. There are no plans to convert to metric at the moment. Since that would be a whole lot of work. But just from what I saw from other people, most parts should work in metric if you take the nearest mm.
(And please ask questions in our discord next time. Git issues are more for bug reports. Thanks)

balzreber, 2020-04-05:
Closed since there is nothing to do here.

closed 2020-04-05 by balzreber