jrsteensen/OpenHornet #185: No attachment between Select Jett Dial and Outer shaft

dnamaster2000, 2020-04-16:
There is no connection between the Select Jettison Dial and the Outer shaft
[image]

dnamaster2000, 2020-04-16:
In my build, i resolved this by creating a bushing to go around the outer shaft, and put a grub screw through the dial and bushing against the outer shaft

balzreber, 2020-04-16:
This is not released yet. So work in progress. I'm sure Noctum will correct that before release.

jrsteensen, 2020-04-20:
Will decide if its an actual issue or is fine with gravity retaining button. Marked for further evaluation and reopened for future tracking.

dnamaster2000, 2020-04-20:
This is related to the outer dial, not the push button

jrsteensen, 2020-05-05:
Looked at the model closer, its a shaft collar with a corrupted model. It's retained.

Fixing it now, and will replace set screw above it with a SHCS so it clocks the button to the dial.

commit 56ad561d (2020-05-05): Fixed SELECT JETT Knob

Closes #185
closed 2020-05-05 by jrsteensen