jrsteensen/OpenHornet #187: Ejection seat pan is too short - by about 5cm

Higgins-OH, 2020-04-17:


jrsteensen, 2020-04-17:
So it appears the seat pan does not need lengthening forwards, but the seat back needs to be pushed aft slightly. Will require greater analysis to figure out the solution.

jrsteensen, 2020-04-17:
So seat pan length should be 18.5 inches. Without a completely redesign, I won't be able to accomplish this. But by shortening the seatback ribs fwd/aft, I should be able to hit about 17.5 inches.

jrsteensen, 2020-04-17:
**Changed parts:**
1. Seat Back Support A
2. Seat Back Support B
3. Seat Pan Cushion
4. Seat Back Cushion
5. Seat Back Top
6. LH Seat Bucket Exterior
7. LH Seat Bucket Exterior Detail
8. RH Seat Bucket Exterior
9. RH Seat Bucket Exterior Detail

commit 1547c997 (2020-04-17): Release Rev C of Seat

INCORPORATED REDLINES AND SHIFTED SEAT BACK AFT TO LENGTHEN SEAT PAN. SEAT CUSHION DIMENSIONS MODIFIED.

Closes #187 & #178
closed 2020-04-17 by jrsteensen