jrsteensen/OpenHornet #192: Flight Stick Electronics

jrsteensen, 2020-04-27:
Need to develop electronics for flight control stick.

Requirements:

1. USB HID I/O for TM Warthog Stick Grip.
2. USB HID I/O for Pitch/Roll Hall Sensors.
3. DCS-BIOS interface for trim motor inputs.
4. NEMA 23 stepper control. 
5. Standard Power Bus connection.
6. USB Out to central Hub.

Questions:
Break into separate modules? (One for stick/gimbal, one for trim control?)

MrMagicMan, 2020-05-04:
I suggest you break it out into separate modules for Stick+Gimbal, and then a separate one for trim, for the people who will not go all out and build the entire thing can build just the stick..

Some questions on the other stuff.

1. USB HID I/O for TM Warthog Stick Grip, 
 Are you referring to reading in the values from the TM grip, if so this will be a valuable resource https://www.reddit.com/r/hoggit/comments/45j5pk/tm_warthog_stick_circuitry/
2. USB HID I/O for Pitch/Roll Hall Sensors.
 Ive done this using a simple arduino sketch, e.g. https://www.instructables.com/id/Create-a-Joystick-Using-the-Arduino-Joystick-Libra/
3. DCS-BIOS interface for trim motor inputs.
-This one is a little more complicated and can be moved to the second module.
4. NEMA 23 stepper control.
-This one is a little more complicated and can be moved to the second module.
5. Standard Power Bus connection.
-Moved to the second module,
6. USB Out to central Hub.
-More of a note on system connections

jrsteensen, 2020-05-05:
I agree. Two modules is probably best.

Yes, will read shift registers from grip, read hall sensors, and that's our stick. Then have an ABSIS module to drive the trim motor outputs and stepper.

@MrMagicMan Any advice on simple 16-bit ADCs we might use with the HID module?

MrMagicMan, 2020-05-06:
I was looking for the AS5600 on the flight control stick assembly but didn't, see it, have we defined the location and layout of key components for the PCBa.

MrMagicMan, 2020-05-06:
https://www.reddit.com/r/hoggit/comments/45j5pk/tm_warthog_stick_circuitry/ this was quite useful

jrsteensen, 2020-05-06:
> I was looking for the AS5600 on the flight control stick assembly but didn't, see it, have we defined the location and layout of key components for the PCBa.

We aren't using an AS5600. We are using two different pot form factor hall sensors. (Same series, different angle range than the ones used on the throttle.

MrMagicMan, 2020-05-06:
Ok so are we using the 6127V1A180L.5 (part numbers don't come across in e drawings) 

Meaning for this board we need to be able to:
Read in the serial data from the stick (modified SPI bus works for this)
Read in 2 12bit ADC's from the hall sensors
Output to 1 USB port as HID joystick, that will supply power etc...

Anything i'm missing?

jrsteensen, 2020-05-06:
Pitch Hall Sensor is 6127V1A90L.5FS
Roll Hall Sensor is 6127V1A360L.5FS

Otherwise correct!

MrMagicMan, 2020-05-07:
Ok that works, I created a quick sketch that creates a hid joystick based on the following library, since I don't actually have a warthog, i also created a quick sketch to emulate the shift registers https://github.com/MHeironimus/ArduinoJoystickLibrary so far no issues found.

jrsteensen, 2020-05-08:
Should we add inputs for the seat controls as well, and consider housing the electronics in the seat pan area?

MrMagicMan, 2020-05-08:
Ideally we want the run for the pots to be as short as possible so they dont pickup noise, as well the shift register will be good at creating set noise. I'm confident that even if we told people not to run the ADC and SR lines together they would and it would become a hassle, is there anyway we can mount this closer to the bottom of the stick then it becomes more of a single unit, we could put the trim stuff in the seat though and have the seat controls go to that board..

jrsteensen, 2020-05-08:
We can do a module directly aft of the gimbal, low to the floor to clear the SSDCTs. Do you have a schematic available? We can do just a small printed Arduino case, with a connector for the stick (was specifying a shielded cable in the interconnect,) and an additional pair of shielded cables from the hall sensors, as well as the USB.

MrMagicMan, 2020-05-08:
I currently just have my hand drawn schematic, the board is going to be pretty small, there are not many components required we can even stick it somewhere on the Pitch Bearing Blocks, or on the front of the main plate if we want and just add a cover. The only connection that is not actually on the stick should be to the USB port. This would make it only one connection off the stick. compared to 5 for the thrustmaster, 6 for the two encoders, and then one more for the USB stick. 

Additional question, are we concerned about the backlash of the gears for the encoder position. I'm concerned we are putting in nice encoders and then maybe losing the gains of this though gear slop.

jrsteensen, 2020-05-08:
It might, depending on actual size. Once you can do a schematic or layout, let me know so I can make the MCAD model and integrate it on the stick. 

As far as gear backlash, the gear ratios are so advantageous that it should be a very very minimal concern.

MrMagicMan, 2020-05-08:
[image]
[image]

MrMagicMan, 2020-05-08:
"schmatic" layout :p

jrsteensen, 2020-05-08:
Haha! It works!

Which arduino are you planning on using for this? I'm a tiny bit confused by that first schematic.

MrMagicMan, 2020-05-08:
Its needs something with native USB (basically anything with the ATmega32u4). The two that are supported without to much work are the Leonardo, and the Micro, I have it running on a pro micro at the moment. 

The first schematic shows the joystick controller on the right (imagine there is another pot on the A0 port, and the fake thrustmaster joystick on the left. That was definitely an internal use only schematic.

MrMagicMan, 2020-05-09:
Additional question, I was looking at the throttle assembly, it looks like there are a bunch of buttons and then there is a couple of analog sensors i'm seeing:

**Analog** (x5)
- Left Throttle 1
- Right Throttle 1
- TDC 2
- Antenna Elevation 1

**Digital**
Approximately 15
- Coms 5
- CageUncage 1
- Speedbrake 2
- Dispense 2
- Exterior Lights 2?
- FLIR 1
- ATC 1
- TDC Depress 1

MrMagicMan, 2020-05-10:
Here is my plan for the throttle assembly. 2 microcontrollers, one in the right throttle and one in the base, inputs from the right throttle go through the shift registers, same strategy as the warthog. The controller in the right throttle will talk to the one in the base through and I2C connection. Schematic should look as follows. 
[image]

jrsteensen, 2021-10-14:
Redesigning roll sensor assy to facilitate the new hall sensor. Will need to gear pitch to get enough of movement to read accurately.

jrsteensen, 2021-10-16:
Done, pending testing/input/redesign for the PCBs.

closed 2021-10-16 by jrsteensen