jrsteensen/OpenHornet #195: Ejection Seat Revision

jrsteensen, 2020-04-30:
Seat pan is still too short. 

Changes required:

Seatpan will extend FWD 3.5".
Vertical rails will have 2.25" trimmed from back of rails.

bjanders, 2020-04-30:
The seat pan base should be extended forward something like this: 
[image]

I eyeballed it to around 1" wide and 4" long and both arcs with a radius about 2". (I made them 20  x 100mm, radius 45mm).

bjanders, 2020-04-30:
If you are making changes then you can also fix this non-issue: https://github.com/jrsteensen/OpenHornet/issues/193. The real benefit is probably that someone does not use time to drill a hole that is not actually needed, or wonder if they are missing something since they can't find a use for it.

jrsteensen, 2020-05-07:
> The seat pan base should be extended forward something like this:
> 
> I eyeballed it to around 1" wide and 4" long and both arcs with a radius about 2". (I made them 20  x 100mm, radius 45mm).
> If you want to be super accurate, then there is screw hole going in from the front into the short edge.

The real kneeguards angle outboard slightly. Adding this with straight kneeguards (necessary due to MDF construction of seat) will only *painfully* affect the user. I probably will not integrate this, though I will consider it.

jrsteensen, 2020-07-02:
Updated issue to reflect complete seat redesign.

uxixu, 2021-01-23:
Might mention that the safe arm handle looks like it has safe on the side too low, instead of towards the edge, as it does in game (DCS).

jrsteensen, 2021-01-23:
@uxixu I'd love to look at it! Please shoot me some pictures, though DCS has enough inaccuracies that it is a reference of last resort on this project.

uxixu, 2021-01-24:
Google Fu not turning up but seems plausible guven the way it hinges.  A safe apears written on the spine, too. Here's from page 78 of Chuck's Guide. I only have Daco fur the Tomcat but if one has the Hornet version may be there in cockpit picture. 

[image]

uxixu, 2021-01-24:
Remembered that F-14D used NACES and DACO shows a couple,  though kinda looks like someone hand wrote safe on the side, though engraved on spine.

[image]

uxixu, 2021-01-24:
That was pilot. Here's RIO. Definitely lower than ED's/Chuck's. 

[image]

jrsteensen, 2021-01-24:
It proabably is, I'm looking for verification on the proper model of ejection seat though - the F18 seat we are creating is a late model SJU-17. A much earlier model was used in the F14, so its not an apple to apples comparison.

jrsteensen, 2021-07-18:
Pushing this to a post-beta release. 

TODO:

- [ ] Rail->Bucket Integration
- [ ] Bucket HW
- [ ] Rail->Pit Integration
- [ ] Add shims under bucket
- [ ] Mount OVRD SW MOUNT
- [ ] Add Seat Bucket Cosmetics
- [ ] Design Seat Placards
- [ ] Add Carbon Fiber Cosmetic Wrap where appropriate.
- [ ] Design headrest/parachute container.

jrsteensen, 2021-09-27:
Pulled this back to beta1 after some significant work on a v1 seat to bring it to 1.5 was discovered. Decided time was better spent on v2 instead. Seat bucket almost done.

jrsteensen, 2021-09-27:
- [ ] Add shims under bucket
- [ ] Add seat decals
- [ ] Add faux rivets to headrest. Add rubber pads where appropriate to headrest.
- [ ] Redesign Drogue Deployment Catapult to allow for installation of catapult and parachute container.
- [ ] Redesign Parachute Deployment Rocket to allow for installation of catapult and parachute container.

jrsteensen, 2021-10-07:
Closed by above reference.

closed 2021-10-07 by jrsteensen