jrsteensen/OpenHornet #202: AMPCD Pot mounting hole wrong size

dnamaster2000, 2020-05-06:
The hole for the pot is the wrong size/shape and missing the key hole on the back, The shell is also too thick, not enough thread passes through to the front for the nut.

[image]

jrsteensen, 2020-05-07:
@dnamaster2000 How thick is the nut that is included with the pot? Its not in the datasheet for the pot.

dnamaster2000, 2020-05-07:
0.092 inches (3/32)

jrsteensen, 2020-05-07:
@dnamaster2000 And hex width, if you have it?

dnamaster2000, 2020-05-07:
.31 across the flats .35 diagonal 

i think the thickest it could be mounted to would be about 1/8th

dnamaster2000, 2020-05-07:
There is also a .401 diameter lock washer .036 thick

jrsteensen, 2020-05-07:
To maintain enough panel thickness, you will have to forgo the lock washer. Otherwise fixed. :) Thanks!

commit 93760d4a (2020-05-07): Updated AMPCD module

Updated pot clearance hole and mating face thickness.

Closes #202
closed 2020-05-07 by jrsteensen