jrsteensen/OpenHornet #203: Hole Mismatch between UIP Bit Box and Bit Box Lid

bnepethomas, 2020-05-07:
There are 4 holes on the near Pilot Lid on bit box Lid (OH1A8A4A-11 - HUD BIT Box Lid.STL) versus just two on the box itself (OH1A8A4A-10 HUD BIT BOX.stl) On Far Right and Mid Right hand side.  And thanks for rescaling the bit box - it works a treat.

jrsteensen, 2020-05-07:
Can you post pictures please?

bnepethomas, 2020-05-08:
[image]
Here you go sir

jrsteensen, 2020-05-08:
This is intentional - there isn't clearance for the screws below it once LEDs and everything is installed. Pretty sure I specified some shorter screws for those two holes for cosmetic purposes.

bnepethomas, 2020-05-08:
cool - I was thinking about having a shallower hole for comestic purposes

jrsteensen, 2020-05-08:
Thanks for bringing it up though! Keep it up!

closed 2020-05-08 by jrsteensen