jrsteensen/OpenHornet #209: Rudder Pedal Interface Plate - mount holes for Solenoid

dnamaster2000, 2020-05-12:
The solenoid has 4 mounting holes, 2 are M3 and 2 are M4 (on each side) the holes on the gantry interface plate are lined up for the M3 holes, but M4 in size.

[image]

dnamaster2000, 2020-05-15:
There may be additional issues with this part, the solenoid is a push type, pushing the other way, needs to be rotated 180, which would leave an issue with retention of the actuator.

jrsteensen, 2020-05-30:
@dnamaster2000 Not sure I'm completing understanding the issues you are seeing. Can you explain a little more in depth with pictures?

dnamaster2000, 2020-05-31:
[image]

The solenoid needs to be mounted in this direction to push the locking arm out of battery, but in this direction the actuator will just fall out the front.  if mounted the other way around, (ignoring the issue of holes in the wrong spot) it wont fit.

The second photo shows even with the locking arm fully to the rear, the solenoid will not be mountable, and would be facing the wrong way.
[image]

jrsteensen, 2020-06-17:
How would you propose we fix it? Any alternate PNs in mind?

jrsteensen, 2020-06-17:
Referenced Wrong Issue...

commit 1821ed02 (2020-06-17): Fixed eccentric nuts

Added new counterbore for eccentric nuts. Doh.
Closes #209
commit 35fab333 (2020-07-01): Fixed Rudder Pedal Gantry Interface Plate

Reversed solenoid
Fixed solenoid mounting screws (located to correct pattern and converted to slots for adjustability.
Lengthened slots for mounting actuator arm
Added retaining cover to solenoid.

Closes #209
closed 2020-07-01 by jrsteensen