jrsteensen/OpenHornet #212: Pushbutton switch MSPS103B (Ldg gear panel) is obsolete

Higgins-OH, 2020-05-23:


L-Walker, 2021-08-01:
Evaluate guarded jett button switch.

jrsteensen, 2021-09-22:
[Also evaluate Greyhill 30-252 BLK](https://www.digikey.com/en/products/detail/grayhill-inc/30-252-BLK/2417135)

Ben-F111, 2021-09-22:
The switch button used for SELECT JETT and RUDDER TRIM works perfect. And fits. I painted the top black. Looks the part and is cheap. 
[image]
[image]

L-Walker, 2021-09-26:
@jrsteensen with your approval, I can use the switch Ben is recommending.

jrsteensen, 2021-09-26:
@L-Walker please do!

L-Walker, 2021-10-09:
Changed as per commit 63b0aae also added a little press fit switch cap to fill the hole and allow it to be painted.
[image]