jrsteensen/OpenHornet #213: Inner grip issues.

MrMagicMan, 2020-05-24:
1. Communication switch has an encoder and is able to rotate. (https://photos.app.goo.gl/aSfpeYJYM2tXSM8q8)
2. Communications switch can rotate in place. and needs significant glue to hold it in place.(https://photos.app.goo.gl/aSfpeYJYM2tXSM8q8)
(1-2) Look at changing to 8 way switch (https://photos.app.goo.gl/j41mbaQUGQoDMsS86)
3. Communications switch cannot be installed without bending pins, the mounting location is in the way (https://photos.app.goo.gl/mjRqkzRxirejwYrR7)
4. Analog stick pot is rubbing on lower TDC Body (
[image])
5. Antenna Elevation switches are too weak (https://photos.app.goo.gl/mjRqkzRxirejwYrR7)

jrsteensen, 2021-10-17:
PCBs TODO:

- [ ] THROTTLE_INNER_GRIP
  - [ ] Select and add LCSC field to metadata for C1, C7, and C8. 
  - [ ] Generate JLCPCB BOM & POS files. 
  - [ ] Any silkscreen work required?
- [ ] ANALOG_STICK_BREAKOUT
  - [ ] Select and add LCSC field to metadata for C1.
  - [ ] Generate JLCPCB BOM & POS files.

jrsteensen, 2021-10-17:
Inner Grip TODO:

- [ ] Integrate the analog_stick_breakout PCB.
- [ ] Integrate the comms switch breakout PCB.
- [ ] Integrate the inner grip breakout PCB.
- [ ] Consider changing comms switch to 8-way hat (mechanically restricted to 4 way + push) or add pin to one side of switch to restrict rotation.
- [ ] Add clearances for analog switch pot.
- [ ] Increase spring forces on TDC springs to overcome friction.