jrsteensen/OpenHornet #215: DDI Buttons are loose feeling

MrMagicMan, 2020-05-27:
I have a feeling there may be too much space beside the buttons, but this may be me.

Video Link
https://photos.app.goo.gl/wrNBT2ha2V4JQwzs6

jrsteensen, 2020-05-27:
Loose how?

MrMagicMan, 2020-05-27:
They can shift around / rotate in the hole, I did not use the specified self tap screw to mount them (it was removed from my Mcmaster order due to anti-dumping regulations. So I put a different button head 4-40 the same length in it, if you look at the end of the video, you can see what im talking about. Also if you want PM me your address and ill send you a board and a button and you can see for yourself.

jrsteensen, 2020-05-27:
In my print of the bezel and buttons, they are tight with a tiny tolerance in them. So I'm a hair confused.

MrMagicMan, 2020-05-27:
Maybe its me then, im talking about the movement you can see at around second 7 of the video.

jrsteensen, 2020-05-27:
We can probably split the tolerance on the button side, but it does risk some potential jamming if people end up with a worst-case MMC button and MMC bezel hole.

MrMagicMan, 2020-05-27:
What if we just make the body for the button holder longer on the DDI, it may stop some of the rotation and may make the button feel better? Do we know if anyone else has tried this i know @balzreber  was getting some domes in soon.

balzreber, 2020-05-31:
At least mine are working just fine. Even if I press on the edge of the button I can't get them to jam. I can upload a vid, if further clarification is needed. And keep in mind that mine are DLP prints, not FDM.

MrMagicMan, 2020-06-01:
I cant get them to jam, they just don't "feel" good (I think they feel cheap, maybe ill play some more with the screw head distance).  I think they will work, i don't know how the real ones feel.

balzreber, 2020-06-01:
Mine feel super solid. No wobble. Nothing. And you don't have that much space to play with the screws. Mine are all the way in, if I start to loosen them, they would jam. Are you testing with just the mechanical parts or do you have the PCB and dome switches all build in?

MrMagicMan, 2020-06-01:
I have everything together (see the first and second image link if you need more details), but my screw head is almost all the way out already, I'm not worried about the device jamming its more that there is play, what screw are you using for this the one in the bom or something else?

balzreber, 2020-06-02:
I used the screws specified.
From your first image it seems as if your PCB is way too deep inside the DDI. Did you mount the back part?
I made a vid of mine. As you can see (and hear) are very tight. No wobble, nothing.
https://share.icloud.com/photos/084So-jRFMqiexsMLvH4gvh-A

MrMagicMan, 2020-06-02:
Looks like I had printed the original DDI Bezel with significantly longer standoffs, ill have to reprint but ill close this issue and open a new one if there are issues. Thanks @balzreber

closed 2020-06-02 by MrMagicMan