jrsteensen/OpenHornet #217: Throttle Assembly Issues and Change Requests

MrMagicMan, 2020-06-01:
After building the throttle here are the issues and some changes I would like.
1. Finger lift retaining screws, I suggest we change this to a shoulder bolt similar to this 90298A141 (although i'm trying to find an equivalent i can source easily, maybe a 3/16" round spacer   92510A694 ), that way we can tighten it. These are the ones in the RH and LH upper adapter, if its not a shoulder bolt then there is no way to torque the bolts in so its loose.
2. Change the 3/16 Nut to a nylock, There's lots of rotation currently and this would stop the nut from loosening. 
3. RH friction space does not attach to anything so its free to rotate and increased tension on the bolt does not increase the throttle resistance.
4. Idle, and Afterburner detent do not have spots for inserts in them, they are threaded right into the plastic..
5. Solenoids are missing from model.
6. Finger lift screws(the ones being held down by springs, should get lock nuts and maybe a a spacer to stop the threads from rubbing and tighten up the lift "feel")
7. Throttle pivot anti rotation bracket is to big for the bolt head, can it be sized to taper in so that multiple bolt head sizes will work.

jrsteensen, 2020-06-30:
> After building the throttle here are the issues and some changes I would like.
>
> 1. Finger lift retaining screws, I suggest we change this to a shoulder bolt similar to this 90298A141 (although i'm [...]
**Agreed.**
> 2. Change the 3/16 Nut to a nylock, There's lots of rotation currently and this would stop the nut from loosening.
**Maybe. Will evaluate if nyloc or loctite is the right answer.**
> 3. RH friction space does not attach to anything so its free to rotate and increased tension on the bolt does not increase the throttle resistance.
**Can you explain this one a little more?**
> 4. Idle, and Afterburner detent do not have spots for inserts in them, they are threaded right into the plastic..
**Detents should be aluminum.**
> 5. Solenoids are missing from model.
**Dang solenoids are gonna be the death of me.**
> 6. Finger lift screws(the ones being held down by springs, should get lock nuts and maybe a a spacer to stop the threads from rubbing and tighten up the lift "feel")
**Agreed. I'll look at smoothing up this mechanism. I am considering a redesign of the fingerlifts to incorporate small alum sheets in places.**
> 7. Throttle pivot anti rotation bracket is to big for the bolt head, can it be sized to taper in so that multiple bolt head sizes will work.
**I'll change it to tighten up the slip fit. Should be an interference fit. Maybe have user heat it and then insert the head into it.**

jrsteensen, 2021-02-19:
@MrMagicMan, I'm starting to look at the throttle so we can finish up the left console. I'm going to avoid any major redesign at this point. I know you built one and submitted a list of changes. I was hoping you can verify that we are looking at the same things so I can get these changes integrated.

> After building the throttle here are the issues and some changes I would like.
>
> 1. Finger lift retaining screws, I suggest we change this to a shoulder bolt similar to this 90298A141 (although i'm [...]

We are talking about this guy?
[image]

> 2. Change the 3/16 Nut to a nylock, There's lots of rotation currently and this would stop the nut from loosening.

Are you talking about the main shaft nuts? (The 3/8th ones?) 

[image]

> 7. Throttle pivot anti rotation bracket is to big for the bolt head, can it be sized to taper in so that multiple bolt head sizes will work.

These guys where the hex head interfaces with them, right?

[image]

I'm going to make minimum changes for now as looking at this design again, I think a few tweaks and it it will be running amazingly.

MrMagicMan, 2021-02-20:
In terms of the comments 1, Yes that guy, 2, Yes but this changed as I made a second iteration see below with images.

I have made two throttles, one based on the initial design and one based on an iteration that both amalgamated a bunch of parts together and changed the main shaft and coupling between the inner and outer throttle (done in the SW2020), (I can't access this right now).

I have been using the second version of the throttle for several months.

Here is the throttle for the side the piece that has the mountings for the finger lifts now connects directly to the grip, and then down to the arm, I removed the small jumper piece. 
[image]

I didn't change the finger lifts, right now a piece of HS tubing is used to space the screw. They have to be loose to work thought this way.
[image]

Changed the connection, to this assembly. uses a hex that goes all the way though, the arms are mounted as large bushings
[image]
this works fine except I didn't grease it so it makes noise. 
The throttle tensions is adjusted by tightening the screw on this side, 
[image]

this pushes the  friction plate via the spring washer against the arm that is sandwiched by it and the center bushing that holds the hex shaft.

[image]

[image]

[image]

[image]
[image]

[image]
[image]

Overall tension can be adjusted by tightening or loosening the bolt on the other side of the tension arm assembly.

B404, 2021-07-24:
Just finished the build of my throttle (Stalk here), with a few things off-spec. 
1. using Metric ISO hardware, except for the throttle pivot 3/8" bolts and bushing hardware. 
2. used OTTO pushbutton switches
3. No solenoids/ATT servos and cut the bottom 100mm and used a different base to fit existing pit frame. 

Using 3mm thick Aluminium for the base plates and throttle arms, I had lateral movement from the arms, causing the grips to strike. To sold this I fitted a spring plunger to the hole below the pivot, and as high as could be. This allows tuning of the lateral position, and a bit of resistance so they don't slop about. 
https://cdn.discordapp.com/attachments/812086922578362368/865881633294254090/IMG_20210717_154335.jpg
Upsizing the pivot bolt may also work, but this seemed to do well AND provide adjustability. 

I also used a 3/8" Locknut on the arm pivot bolt. It alows me to adjust the tension on the arms, especially where the friction arm sits. I have noticed it still moved with the RH throttle arm occasionally. 

Grips: The LH Grip printed and came together beautifully. The RH one had a few small issues. 
I suggest adding a screw at the top of the grip, from just in front of the Dispense switch, rather than just a dowel. Help to pull it together with the other half, which cannot have a screw hole through there. 

Also check that the 4-way Comms switch can be fitted in the space. Could be printing error, but I had to melt a way in. That switch also appears to close contacts before it 'clicks'. 

Still not happy with the feel and behaviour of the Ant Elev. control, as I can't get the exact springs or pot. Just as cost effective for me to use an OTTO RTC thumbwheel, so will document any changes to make that work.

B404, 2021-07-24:
> > After building the throttle here are the issues and some changes I would like.
> > 
> > 2. Change the 3/16 Nut to a nylock, There's lots of rotation currently and this would stop the nut from loosening.
> >    **Maybe. Will evaluate if nyloc or loctite is the right answer.**
Nylock. Allows you to set the compression on the arms, etc 'just' so. 

> > 3. RH friction space does not attach to anything so its free to rotate and increased tension on the bolt does not increase the throttle resistance.
> >    **Can you explain this one a little more?**
Tightening the nut just jams the friction arm, spacer and throttle arm together, moving as one. If you set it loose enough for the friction arm to be free to move independently of the throttle arm, then the throttle arm just flops about. (inline, but see my note about lateral - that also fixed this to a degree.

> > 4. Idle, and Afterburner detent do not have spots for inserts in them, they are threaded right into the plastic..
> >    **Detents should be aluminum.**
Plastic works, just upsize the screw and hand tap the thread into them. 

> > 7. Throttle pivot anti rotation bracket is to big for the bolt head, can it be sized to taper in so that multiple bolt head sizes will work.
> >    **I'll change it to tighten up the slip fit. Should be an interference fit. Maybe have user heat it and then insert the head into it.**
I did not have this issue. 8.8 3/8" bolt from my garage worked fine.

jrsteensen, 2021-09-13:
Evaluate using electromagnets for detents vs solenoids.