jrsteensen/OpenHornet #221: Rudder Pedal Interface Plate - Holes for eccentric nuts

dnamaster2000, 2020-06-13:
The holes for the eccentric nuts are not the proper size for the nuts selected. they are only used as spacers in the current arrangement. 
[image]

jrsteensen, 2020-06-13:
Did you make the holes counterbored? (They should show as counterbored holes in the model.)

dnamaster2000, 2020-06-13:
Part was 3D printed, the through hole is M5 diamater, and the nuts sit in the counterbore, but doesn't provide any adjustment, the hole should be a through hole, with the diameter of the eccentric nut base, so that when the nut is rotated, it effectively moves the center, allowing the spacing (tension) of the wheel to be adjusted. good visuals here: https://www.youtube.com/watch?v=uMes1My5y5I

jrsteensen, 2020-06-17:
[image]

I'm not totally sure how/why this doesn't work? The bore is constrained with a counterbore on bottom to allow it to turn at the correct height to ensure V-Wheel/Rail engagement, and a counterbore on the top for the SHCS.

dnamaster2000, 2020-06-17:
it looks/sounds like the eccentric nuts are not being used to adjust the position of the wheel, if that's the case, these wheels should be setup the same way as the opposing ones, using just a spacer.

jrsteensen, 2020-06-17:
No, if installed as shown, with the correct barbell shaped counterbore, it will adjust the position of the wheel horizontally. If the issue is the adjustment hex is interfering with the plate, we can open up the bottom counterbore to eliminate the interference.

jrsteensen, 2020-06-17:
....I'm an idiot.

Before: 
[image]

After:
[image]

jrsteensen, 2020-06-17:
Closed in  1821ed0.

closed 2020-06-17 by jrsteensen