jrsteensen/OpenHornet #235: Caution light and J1 issue

Kerube, 2020-07-12:
issue with the Caution lights and the PCB, J1 lands on top of the face plate. ended up having to sand down the connector pins to make it work.

jrsteensen, 2020-07-12:
Did you use the new ones from a couple weeks ago? I increased the openings for connectors.

Kerube, 2020-07-12:
ah, nope. sorry.

jrsteensen, 2020-07-12:
As you were, I was thinking of EWI. Can you provide some measurements/location/detail pics of the issue so I can update the model?

Kerube, 2020-07-13:
[
[image]
](url)

Kerube, 2020-07-13:
here is the image for the Caution Advisory and in red are J1 pins and the Indicator case.

L-Walker, 2021-01-30:
Just to update this, I will be creating a replacement PCB for this using the RGB pixel controllable backlighting system being deployed everywhere else in the pit, I intend to place the connectors saddling each side of a divider alleviating this issue. There will be 2 Molex Mini-Fit connectors.

closed 2021-02-02 by jrsteensen