jrsteensen/OpenHornet #237: JETT panel PCB orientation and parts size

Kerube, 2020-07-18:
Found an issue with JETT Panel PCB. is oriented wrong.
And a small request if possible. all the components seem to be 402 footprint
Can they be changed to 602?
After buying so many 602 parts it would not make sense to change to 402
[image]

jrsteensen, 2020-07-18:
Thanks for the report - I just verified this issue in ECAD.

As far as request for 0602, unfortunately, we aren't able to physically accommodate that on this PCB. 

During the redesign, it should be designed so the front (LEDs) can be assembled by jlcpcb as a requirement.

L-Walker, 2021-02-02:
This issue is now redundant due to this PCB being replaced with one based on the Backlighting system.

closed 2021-02-02 by jrsteensen