jrsteensen/OpenHornet #245: INT lights backplate cutouts for pots missing

Higgins-OH, 2020-08-05:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
The used pots need a small cutout to prevent turning.

### Expected Results/How It Should Work:
Small cutouts for the nubs.

### Actual Results/How It Does Work:
No holes.

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*
INT lights backplate

L-Walker, 2021-01-30:
For the record this is incorrectly labeled INT lights panel, it actually refers to the EXT Lights Panel on the right console.

Higgins-OH, 2021-02-01:
Thank you for noticing!

L-Walker, 2021-02-02:
@jrsteensen The changes to complete this issue are included in commit [d0a675c](https://github.com/jrsteensen/OpenHornet/commit/d0a675c2168bb1c3cb62f6e4a84cc7e34bb45d3d)
Completion of this issue is available upon your approval.
[image]

closed 2021-02-02 by jrsteensen