jrsteensen/OpenHornet #254: Add MFD displays to the Main Instrument Panel

JKI757, 2020-08-29:
### Summary:
Multi-Function Displays

### Expected Results/How It Should Work:
Was looking around on your repo to see if you'd considered the MFD issue, as this is one thing it seems nobody has really been able to find good displays for.  
Just recently based off a reddit post I was able to find a supplier in Shenzhen named Shanghai Top Display OptoElectronics that makes a 4" display that is not perfect, but does appear to be suitable for this.  
I'm currently working on two things -- adding in the additional hardware required to make those connect (will require either HDMI connections from the graphics card, or usb-HDMI dongles, which I believe is the better way to go (I purchased usb-hdmi dongles, but am having difficulty making them work with Windows 10 -- I used a different USB-C dongle with a Mac to test the displays as you can see below in the picture).  I have not addressed any large scale purchasing from them, currently have purchased three sample 4" displays, which they shipped to me within about a week.  These are small within the MFD frame, so I will in the future address with them getting 5" displays (they don't offer those on their website, so they may not be open to designing them unless we can get a larger order put together).  The total cost for these displays was $48 for the hdmi-mpi board + $16 for the display itself + 20 bucks shipping for the three I ordered.  These are samples, so a contract for larger numbers could probably be negotiated to drop the cost significantly if there were interest eventually.

[image]

### Actual Results/How It Does Work:
Multi-Function Displays are required to operate the aircraft, they allow managing stores, display moving map, HUD, targeting information from TGP, Maverick missile targeting, HSI, etc.

### Screenshots:

![Dispay in Action](https://user-images.githubusercontent.com/27347681/91638612-b4969380-e9de-11ea-940a-
daa461bfb908.jpeg)

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [X] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):

jrsteensen, 2020-08-29:
I appreciate your research, but all the internal displays have already been selected, utilizing commonly available COTS displays. The part numbers are the name of the display models in the UIP assembly. I'm going to close this issue.

Thanks for taking the time to post though!

closed 2020-08-29 by jrsteensen

JKI757, 2020-08-29:
Maybe I'm really missing something, but I've searched high and low through every branch of this repo and I simply don't see a part number for the display you're using.  I'm really interested in this, can you point me to a link or something?