jrsteensen/OpenHornet #256: Rotate Hyd Preassure bezzle 45*

dnamaster2000, 2020-10-23:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
Attachment screws for Hyd Preassure guage are obstructed by other items on the panel, rotating 45* will resolve, and not sure about real hornets, but will match DCS hornet

### Expected Results/How It Should Work:
see picture

### Actual Results/How It Does Work:
see picture

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X ] Bug
- [X ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

L-Walker, 2021-10-09:
This should be tackled at the same time as doing issue #322

jrsteensen, 2021-10-31:
CLosed in referenced commit.

commit cb6b7c4c (2021-10-31): Rotated HYD PRESS Indicator 45*

Closes #256
closed 2021-10-31 by jrsteensen