jrsteensen/OpenHornet #257: Dual Needle Gauge Servo change

dnamaster2000, 2020-10-23:
### Summary:
The servo's selected for the gauges are very difficult to find, could it be changed to SG90 Micro Servo like these, they are very cheap, and easily found, as they are used in a lot of RC builds.  amazon search here shows many: https://www.amazon.ca/s?k=SG90&ref=nb_sb_noss_2 

these could also be used in-place of the VID-19 for single needle gauges as well, they are easier to program for and faster operating. 

### Expected Results/How It Should Work:
Functions the same way, every slightly larger

### Actual Results/How It Does Work:
Functions the same way, every slightly larger

### Screenshots:
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [] Left Console
- [X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2020-10-24:
I'd require an accurate CAD model to include servo spline to proceed with this. This will only be used in the dual needle gauges - the VID-19 has Arduino libraries in common use and are the defacto standard for single needle gauges.

Ben-F111, 2021-06-08:
I have made these using the below SERVO's I have updated my own Fusion (STEP) Files, and proven they work

https://core-electronics.com.au/sub-micro-servo-3-7g-generic.html

jrsteensen, 2021-08-20:
I'm pretty sure thats the one we specified @Ben-F111. (Please correct me if I'm wrong!) Its manufacturered and sold here on this pond by Pololu.

https://www.pololu.com/product/1053

jrsteensen, 2021-10-29:
Pausing this (and giant gears!) while I evaluate the X40 8798 dual shaft stepper.

https://www.jukenswisstech.com/JSTFiles/downloads/2011/03/X40_Flyer_v1.1.pdf

jrsteensen, 2021-11-07:
Closed in referenced commit (eliminated servos, using X40 dual needle microstepper.)

closed 2021-11-07 by jrsteensen