jrsteensen/OpenHornet #265: Electronics Redesign

jrsteensen, 2020-10-29:
**This is a placeholder issue that will be broken out into many other issues**

### Summary:
Redesign OpenHornet electronics stack.

### Goals:
- Simplify construction by having all SMD components on one side of PCB.
- All SMD components should be stocked by JLCPCB to facilitate delivery with all SMD components pre-assembled. (PTH components to be populated by user.)
- Add test points as required to PCB.
- Eliminate all currently used LEDs. Utilize ONLY either the WS2812B-Mini or the WS2812B-B LEDs. (Individually addressible and capable of being driven from a single Arduino PWM pin + PWR per string.)
- Make ABSIS boards more general, with feature specific small breakout PCBs.
- Panelize PCBs to drive component quantity discounts.
- Remove dome switches from pit. Replace with quality tactile switch. Optimize for backlighting.
- Support backlighting the OpenHornet pit.

## Issues/Actions:
- [ ] Create detailed OpenHornet IO requirements table.
- [ ] Establish common electronics parts to be used across all PCBs. (JLCPCB standard list parts preferred, followed by extended list parts.)
- [ ] Create detailed electronic requirements document.

### General PCBs:
- [ ] ABSIS Nano
- [ ] ABSIS Mega
- [ ] ABSIS Bus Master(?)
- [ ] 12V Output Drivers

### UIP PCBs:
- [ ] .55 SQ Indicator PCB
- [ ] Master Mode Pushbutton PCB
- [ ] UFCv4
- [ ] DDI/AMPCD Controller PCB
- [ ] DDI Button PCB
- [ ] AP Module PCB
- [ ] Right Eyebrow Warning Indicator PCB
- [ ] Left Eyebrow Warning Indicator PCB

### LIP PCBs:
- [ ] AMPCD Button PCB
- [ ] IFEI Button PCB
- [ ] JETT Select PCB
- [ ] RWR Control Panel PCB

### Right Console PCBs:
- [ ] Caution Panel PCB

### Left Console PCBs:
- [ ] Throttle PCB(s?)

- [ ] Update Cockpit Interconnect Drawing

dnamaster2000, 2020-10-31:
A note on the DDI / AMPCD Button Boards, Pull Up resistors are required on all input pins for the TCA9534 to work correctly.

petersawka, 2020-10-31:
Working on adding the Pull-up Resistors on the DDI and AMPCD Button Boards.

jrsteensen, 2021-09-10:
This issue is complete. If any more functional PCBs are required, they will be completed as we progress towards hardware freeze.

closed 2021-09-10 by jrsteensen