jrsteensen/OpenHornet #281: Parking Brake - Extra Detent

dnamaster2000, 2020-12-30:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
Due to the through hole on the screw that blocks rotation, there is a second detent in the Emergency position of the lever, you can see it in the video posted below. 

### Expected Results/How It Should Work:
(Explain how you think it should work here. Be very specific.)

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.

https://www.youtube.com/watch?v=n1THPwVa9PM

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2020-12-30:
I will evaluate blind hole/shorter screw for this position. The other alternative would be through hole, with a shorter screw, then pot the hole with JB weld, and wipe the exterior smooth with a paper towel or something prior to the JB weld curing.

dnamaster2000, 2021-01-04:
I think you can remove the back detent entirely, add an additional 2 holes to the shaft at front detents for when the handle is stowed. would probably give a more positive detent on rotate, and remove redundancy.

dnamaster2000, 2021-01-16:
[image]

If you remove the detent at the aft (#1) in my picture, you can put a second set of holes on the forward detent to take care of the pivot action (#2) in my picture.  this actually results in less overall parts for the entire mech.  also, the holes for detents do not need to be through holes, making them 1/16th or so in depth will increase overall strength of the rod, and provide the same function.

Ben-F111, 2021-06-15:
Park Brake shaft enhancement #302 I have suggested a full diameter grove be used, this functions just as good, and is very easy to make compared with getting the 2 x holes drilled

jrsteensen, 2021-07-26:
Converted 4-40 holes to .1875 blind holes, and replaced 91251A108 SHCS with 91251A106 SHCS, in addition to incorporating #302.

[image]

commit 745b34f7 (2021-07-26): Parking Brake Mech Bug Fixes

Parking Brake Shaft: Converted 4-40 thru holes to .1875 blind holes, and replaced 91251A108 SHCS with 91251A106 SHCS. (Resolves #281)
Added .0625 of material to outside wall of parking brake mechanism body. (Resolves #280)
closed 2021-07-26 by jrsteensen