jrsteensen/OpenHornet #286: Mag Held switch drill size and position to change.

B404, 2021-01-07:
### Summary:
Following dimensions from the 3D model, the drill size and position was too large and low to clear internal components not included in the APEM 3D model. 

### Expected Results/How It Should Work:
Drill in right spot should clear the internal contacts. 

### Actual Results/How It Does Work:
Internal moving part fouls the retraction of the pin/dowel. 

### Screenshots:
Hole centre location change. 
[image]

Hole bore size to change.
[image]

2.5mm Drill bit *just* clears the contact. 
[image]

## More Information
Not sure if the way it should be - but as pictured (drilling from the '3' end of the switch body, the 631H2 is open circuit when the toggle would be indicating 'Arm'  on LTD/R. 

Parts purchased are APEM 631H/2 SPST NKL On-None-Off. PN on Mouser.com is 642-632H2, for reference. 

### Category:
*Check one or more items.*
- [X ] MCAD (SolidWorks)
- [X ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X ] Left Console
- [X ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
Ah.

Ben-F111, 2021-07-11:
This Mag switch design may not work correctly, The current design is for a PUSH off scenario, a MAG SW is a Hold on system
THe Spec'd Plunger is also not strong enough to push the switch to the off position

jrsteensen, 2021-10-31:
Switch Name | Bat | Function | Mag Held Position
-- | -- | -- | --
Launch Bar | Standard 17.5mm   bat, milled flat and Drill/Tapped (M2.5 X 0.45) | ON-(ON) | EXTEND
Hook Bypass | Standard 17.5mm bat | ON-(ON) | FIELD
APU | Standard 17.5mm bat   (Round locking Toggle Bat will press fit on) | ON-(ON) | ON
ENG CRANK | Standard 17.5mm bat | (ON)-OFF-(ON) | L & R
DUMP | Standard 17.5mm bat   (Round locking Toggle Bat will press fit on) | ON-(ON) | ON
PITOT | Standard 17.5mm bat   (Round locking Toggle Bat will press fit on) | ON-(ON) | ON
LTD/R | Standard 17.5mm bat   (Round locking Toggle Bat will press fit on) | ON-(ON) | ARM
Canopy | Standard 17.5mm bat | (ON)-OFF-(ON) | OPEN (ONLY)

jrsteensen, 2021-11-19:
Initial testing by @Ben-F111 using the Xurui XT-23AF switches and 2x electromagnets are extremely promising. 

Required dimensions:

- [ ] Switch Body
- [ ] Switch Bat
- [ ] Switch bushing
- [ ] Location and size of slot in body
- [ ] Hole diameter for shaft.
- [ ] Dimensions of EM mounting bracket (angle and EM mounting hole location.)

jrsteensen, 2022-03-29:
Ordered an APEM 645H/2 to teardown and see if we can't give it the same treatment. The APEMs tend to have a less mushy feel to them compared to the Xurui switches.

jrsteensen, 2022-04-07:
Added all new mag switches to relevant assemblies, and have the -1 thru -3 variants mostly designed.

- [ ] Add actuator to -1
- [ ] Add actuator to -2
- [ ] Add actuator to -3
- [ ] Redesign canopy box to accommodate mag switch.

jrsteensen, 2022-04-10:
Issue closed with the commit of OHC0005-1 thru -3 to F360 Repository.

Woohoo! 

[image]

closed 2022-04-10 by jrsteensen