jrsteensen/OpenHornet #288: Landing Gear inner shaft

dnamaster2000, 2021-01-16:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
I ran into a couple of issues:
1) There is excessive space around the rear portion of the axis, this results in alot of movement, and with the detent installed, the shaft does not sit straight. 
2) The detent tip is nylon, and has a poor interface with the 3d printed track.  on the first attempt to use this i broke the detent tip from over rotating (it sticks then lets go) even with a second detent, i was unable to 'feel' the track.  my 3D printed parts were DLP prints, the resin is very smooth, i woudl imagine if it was a FDM part it would have been worse.
3) there is no way to remove the detent once it is seated in the groove at present.

### Expected Results/How It Should Work:
unknown

### Actual Results/How It Does Work:
see summary

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ X ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ X ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

Ben-F111, 2021-06-15:
I have suggested via a suggestion via Landing Gear adjustment hole required #306 to add a hole to allow the ball spring to be set tot the correct depth, when it is the landing gear does function good

jrsteensen, 2021-09-22:
Addressed in #306. Thanks for the report!

closed 2021-09-22 by jrsteensen