jrsteensen/OpenHornet #301: Oxy Flow is a Potentiometer in DCS BIOS

bnepethomas, 2021-03-12:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
(Please summarize your issue here. Be specific.)

### Expected Results/How It Should Work:
In both 'original' and Flight Panels DCS BIOS - the Oxy Flow is a potentiometer
### Actual Results/How It Does Work:
Ref https://discord.com/channels/392833351238811648/392835164704604160/454145166828109827

### Screenshots:
                                                                     },
                                                         "OXY_FLOW": {
                                                                             "category": "LOX Indicator",
                                                                         "control_type": "limited_dial",
                                                                          "description": "OXY Flow Knob",
                                                                           "identifier": "OXY_FLOW",
                                                                               "inputs": [ {
                                                                                             "description": "set the position of the dial",
                                                                                               "interface": "set_state",
                                                                                               "max_value": 65535
                                                                                         }, {
                                                                                                "description": "turn the dial left or right",
                                                                                                  "interface": "variable_step",
                                                                                                  "max_value": 65535,
                                                                                             "suggested_step": 3200

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ X] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [X ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [X] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-03-12:
This should probably be changed in DCS BIOS. Functionality is on and off IRL; there is no need for an analog input.

bnepethomas, 2021-03-12:
Cool - it makes sense - use map function as Ben suggested

closed 2021-03-12 by bnepethomas