jrsteensen/OpenHornet #304: Add Electrically held switch electronics for Canopy Switch

bnepethomas, 2021-04-04:
### Summary:
The current ABSIS branch (2.0) is missing a port for the Magnetic Canopy switch

### Expected Results/How It Should Work:
Additional output port needed

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [ X] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [X ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-04-04:
How is it missing? Thats what the Solenoid Accessory Driver PCB is for: ECAD\PCBs\ABSIS\ABSIS_Solenoid_Driver_Accessory

bnepethomas, 2021-04-04:
Hi John - I could be looking at the wrong thing - I was referencing 
https://github.com/jrsteensen/OpenHornet/blob/ABSIS_2.0/ECAD/docs/OpenHornet%20Block%20Diagram.pdf
Which doesn't include the canopy held switch

jrsteensen, 2021-04-04:
You are correct there. Since we don't have a two way mag held switch design, we weren't supporting it. (Using a three position momentary toggle instead). It would be pretty straight forward to route the solenoids of the mag switch to one of the existing accessory PCBs instead for the time being.

bnepethomas, 2021-04-04:
Agree - just ensuring its added to the thought pile as its much easier to include it now rather than later

closed 2021-04-04 by bnepethomas