jrsteensen/OpenHornet #305: JLCPCB rejects UFC display layers

Alan-JH, 2021-05-19:
### Summary:
JLCPCB rejected my order with display layers 2-4 for opt, com, and scratchpad, citing the minimum allowed clearance between routed cutouts being 4mm, while the PCBs have a clearance of about .32mm. 

### Expected Results/How It Should Work:
Expected JLC to process the order, and some number of pieces to fail

### Actual Results/How It Does Work:
JLC rejected the order before manufacturing.

### Screenshots:

[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ ] MCAD (SolidWorks)
- [x] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ ] Feature Enhancement
- [x] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [x] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):

ufc_v4_comm_layer2.kicad_pcb
ufc_v4_comm_layer3.kicad_pcb
ufc_v4_comm_layer4.kicad_pcb
ufc_v4_opt_layer2.kicad_pcb
ufc_v4_opt_layer3.kicad_pcb
ufc_v4_opt_layer4.kicad_pcb
ufc_v4_scratchpad_layer2.kicad_pcb
ufc_v4_scratchpad_layer3.kicad_pcb
ufc_v4_scratchpad_layer4.kicad_pcb

jrsteensen, 2021-08-27:
That would be great! I'm curious if clear resin will do the trick or if we will need to add a tiny bit of white powder to give the light something to reflect off of. Did you have to modify the layers to print them with a .2mm nozzle?

Alan-JH, 2021-08-27:
It turns out the clear resin I have went bad. I could get some more soon, but for now I tried white glue instead, as I have used that to diffuse LEDs before. 

I had to close the slots for the 7 segment digits on the scratchpad by 0.05mm to get the slicer to slice the model properly with a 0.2mm nozzle. Otherwise, it didn't need modification. The nozzle clogged every 4 prints or so though, so I would either need to cold pull or replace it entirely, which wasn't pleasant, hence my suggestion for SLA.

Alan-JH, 2021-08-28:
After drying overnight, the glue definitely helped, but blocks a little too much light. I will try to get some clear resin soon and give that a shot

jrsteensen, 2021-09-04:
Resin print of layers 2 and 3 were successful as-is, or can be FDM printed with a .2 nozzle. No change required to model. Will note for them to be printed on drawing.

closed 2021-09-04 by jrsteensen