jrsteensen/OpenHornet #310: Alignment of Ribs

Lazarus94, 2021-07-10:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
(Please summarize your issue here. Be specific.)
There is very little tolerance for the alignment of the ribs in the Left/Right Console, Upper Instrument Panel, and Lower Instrument Panel. By hand, it is difficult to perfectly align ribs. If the ribs are not near-perfectly aligned there are issues with inserting printed brackets and other parts. 

### Expected Results/How It Should Work:
(Explain how you think it should work here. Be very specific.)

In order to solve this, I propose adding grooves to floor plates where ribs for example are mounted. Whilst there are holes in the floor panels for where the screws must come through it is very difficult to perfectly align the ribs with the holes. By adding grooves for where the ribs should fit in it would make the assembly greatly more simple and easy.

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

It works by just adding a groove into the floor plates of the consoles and UIP/LIP. Then the builder doesn't have to worry about proper alignment. They can just put the parts into the groove and screw it in. 

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*
[image]
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [X ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [ X] Lower Instrument Panel (LIP)
- [ X] Main Instrument Panel (MIP)
- [X ] Left Console
- [ X] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*
OH1A-100
Right Console Floor
Left Console Floor

jrsteensen, 2021-08-01:
- [ ] Add 2x 2040 extrusion to each console and LIP. 
- [ ] Add cutouts on each rib to clearance the rails.
- [ ] Add 90 degree brackets and hardware to mount the rails at the bottom to the 2040 extrusion.

jrsteensen, 2022-08-10:
Consider eliminating edge drills and introducing tolerance by using combination of corner brackets and triangle corner brackets with floating or fixed bolted joints.