jrsteensen/OpenHornet #311: KY58 Panel - Rotary Switch - A12505RNZQ

Ben-F111, 2021-07-11:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
KY58 Panel - Rotary Switch - A12505RNZQ - Incorrect Part Number

### Expected Results/How It Should Work:
KY58 Panel - Rotary Switch - A12505RNZQ - Is a 8 Pole 45 Deg Switch, should be a 36 Deg Switch to align with the Text

Potential replacement Part Number C&K R10815RS02Q

### Actual Results/How It Does Work:
The 45 Deg Switch does not align with the text

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] ECAD (PCB Design or other electrical hardware)

### Type:
*Check one item.*
- [X] Bug

### Applicable End Item:
*Check one item.*
- [X] Right Console

### Associated Filename(s):
N/A

jrsteensen, 2021-10-31:
Requested model of 71BD36-01-1-AJN from Grayhill.

jrsteensen, 2021-11-04:
Closed in referenced issue.

closed 2021-11-04 by jrsteensen