jrsteensen/OpenHornet #313: v1.0.0-beta1 Drawing Package

jrsteensen, 2021-07-23:
## UIP
- [ ] Mechanical Drawings
  - [ ] OH1A1 - ASSY, UPPER INSTRUMENT PANEL
  - [ ] OH1A2 - ASSY, PANEL, MASTER ARM
  - [ ] OH1A3 - ASSY, DDI
  - [ ] OH1A4 - ASSY, EYEBROW WARNING INDICATOR, LEFT
  - [ ] OH1A5 - ASSY, EYEBROW WARNING INDICATOR, RIGHT
  - [ ] OH1A6 - ASSY, PANEL, SPIN RECOVERY
  - [ ] OH1A8 - ASSY, UPFRONT CONTROL PANEL
    - [ ] OH1A8A1 - ASSY, UPFRONT CONTROL PANEL PRIMARY MODULE
    - [ ] OH1A8A2 - ASSY, PANEL, HUD
    - [ ] OH1A8A3 - ASSY, AP SWITCH MODULE
    - [ ] OH1A8A4 - ASSY, HUD BIT BOX
- [ ] Add items to Interconnect Drawings
  - [ ] OH1A1 - ASSY, UPPER INSTRUMENT PANEL
  - [ ] OH1A2 - ASSY, PANEL, MASTER ARM
  - [ ] OH1A3 - ASSY, DDI
  - [ ] OH1A4 - ASSY, EYEBROW WARNING INDICATOR, LEFT
  - [ ] OH1A5 - ASSY, EYEBROW WARNING INDICATOR, RIGHT
  - [ ] OH1A6 - ASSY, PANEL, SPIN RECOVERY
  - [ ] OH1A8 - ASSY, UPFRONT CONTROL PANEL
    - [ ] OH1A8A1 - ASSY, UPFRONT CONTROL PANEL PRIMARY MODULE
    - [ ] OH1A8A2 - ASSY, PANEL, HUD
    - [ ] OH1A8A3 - ASSY, AP SWITCH MODULE
    - [ ] OH1A8A4 - ASSY, HUD BIT BOX
## LIP
- [ ] Mechanical Drawings
    - [ ] OH2A1 - ASSY, LOWER INSTRUMENT PANEL
    - [ ] OH2A1A1 - ASSY, LIP STRUCTURE
    - [ ] OH2A2A1 - ASSY, LEFT LIP MODULE
        - [ ] OH2A2A1A1 - ASSY, PANEL, JETT SLEECT
        - [ ] OH2A2A1A2 - ASSY, PANEL, IFEI
        - [ ] OH2A2A1A2 - ASSY, PANEL, VIDEO RECORD
    - [ ] OH2A3A1 - ASSY, AMPCD MODULE
    - [ ] OH2A4A1 - ASSY, PANEL, RWR CONTROL
    - [ ] OH2A5A1 - ASSY, PANEL, ECM/DISP
    - [ ] OH2A6A1 - ASSY, CABIN PRESSURE GAUGE
    - [ ] OH2A7A1 - ASSY, STANDBY INSTRUMENT MODULE
- [ ] Add items to Interconnect Drawings
    - [ ] OH2A1 - ASSY, LOWER INSTRUMENT PANEL
    - [ ] OH2A1A1 - ASSY, LIP STRUCTURE
    - [ ] OH2A2A1 - ASSY, LEFT LIP MODULE
        - [ ] OH2A2A1A1 - ASSY, PANEL, JETT SLEECT
        - [ ] OH2A2A1A2 - ASSY, PANEL, IFEI
        - [ ] OH2A2A1A2 - ASSY, PANEL, VIDEO RECORD
    - [ ] OH2A3A1 - ASSY, AMPCD MODULE
    - [ ] OH2A4A1 - ASSY, PANEL, RWR CONTROL
    - [ ] OH2A5A1 - ASSY, PANEL, ECM/DISP
    - [ ] OH2A6A1 - ASSY, CABIN PRESSURE GAUGE
    - [ ] OH2A7A1 - ASSY, STANDBY INSTRUMENT MODULE
## L CONSOLE
- [ ] Mechanical Drawings
    - [ ] OH4A1 - ASSY, LEFT CONSOLE
    - [ ] OH4A1A1 - ASSY, LEFT CONSOLE STRUCTURE
    - [ ] OH4A2A1 - ASSY, LANDING GEAR PANEL
    - [ ] OH4A3A1 - ASSY, SELECT JETT PANEL
    - [ ] OH4A3A2 - ASSY, PARK BRAKE HANDLE
    - [ ] OH4A3A3 - ASSY, BRAKE PRESS GAUGE
    - [ ] OH4A4A1 - ASSY, FIRE TEST PANEL
    - [ ] OH4A4A2 - ASSY, EXTERNAL LT PANEL
    - [ ] OH4A4A3 - ASSY, GEN TIE PANEL
    - [ ] OH4A5A1 - ASSY, FUEL PANEL
    - [ ] OH4A5A2 - ASSY, APU/ENG CRANK PANEL
    - [ ] OH4A6A1 - ASSY, FCS PANEL
    - [ ] OH4A7A1 - ASSY, COMM PANEL
    - [ ] OH4A7A2 - ASSY, OBOGS PANEL
    - [ ] OH4A7A3 - ASSY, MC/HYD PANEL
    - [ ] OH4A7A4 - ASSY, ANT SEL PANEL
    - [ ] OH4A8A1 - ASSY, LEFT CIRCUIT BREAKERS
    - [ ] OH4A8A2 - ASSY, DISPENSE BUTTON
    - [ ] OH4A8A3 - ASSY, NUC ENABLE SWITCH
    - [ ] OH10A1A1 - ASSY, SIM PWR PANEL
- [ ] Add items to Interconnect Drawings
    - [ ] OH4A1 - ASSY, LEFT CONSOLE
    - [ ] OH4A1A1 - ASSY, LEFT CONSOLE STRUCTURE
    - [ ] OH4A2A1 - ASSY, LANDING GEAR PANEL
    - [ ] OH4A3A1 - ASSY, SELECT JETT PANEL
    - [ ] OH4A3A2 - ASSY, PARK BRAKE HANDLE
    - [ ] OH4A3A3 - ASSY, BRAKE PRESS GAUGE
    - [ ] OH4A4A1 - ASSY, FIRE TEST PANEL
    - [ ] OH4A4A2 - ASSY, EXTERNAL LT PANEL
    - [ ] OH4A4A3 - ASSY, GEN TIE PANEL
    - [ ] OH4A5A1 - ASSY, FUEL PANEL
    - [ ] OH4A5A2 - ASSY, APU/ENG CRANK PANEL
    - [ ] OH4A6A1 - ASSY, FCS PANEL
    - [ ] OH4A7A1 - ASSY, COMM PANEL
    - [ ] OH4A7A2 - ASSY, OBOGS PANEL
    - [ ] OH4A7A3 - ASSY, MC/HYD PANEL
    - [ ] OH4A7A4 - ASSY, ANT SEL PANEL
    - [ ] OH4A8A1 - ASSY, LEFT CIRCUIT BREAKERS
    - [ ] OH4A8A2 - ASSY, DISPENSE BUTTON
    - [ ] OH4A8A3 - ASSY, NUC ENABLE SWITCH
    - [ ] OH10A1A1 - ASSY, SIM PWR PANEL
## R CONSOLE
- [ ] Mechanical Drawings
    - [ ] OH5A1 - ASSY, RIGHT CONSOLE
    - [ ] OH5A1A1 - ASSY, RIGHT CONSOLE STRUCTURE
    - [ ] OH5A2A1 - ASSY, HOOK LEVER & INDICATOR
    - [ ] OH5A2A2 - ASSY, WING FOLD HANDLE
    - [ ] OH5A2A3 - ASSY, AV COOL SWITCH
    - [ ] OH5A2A4 - ASSY, RADAR ALTIMETER GAUGE
    - [ ] OH5A2A5 - ASSY, HYD PRESS GAUGE
    - [ ] OH5A3A1 - ASSY, CAUTION LIGHT PANEL
    - [ ] OH5A4A1 - ASSY, ELEC PANEL
    - [ ] OH5A5A1 - ASSY, ECS PANEL
    - [ ] OH5A5A2 - ASSY, DEFOG/WINDSHIELD PANEL
    - [ ] OH5A6A1 - ASSY, INTR LT PANEL
    - [ ] OH5A7A1 - ASSY, SNSR PANEL
    - [ ] OH5A8A1 - ASSY, SIM CNTL PANEL
    - [ ] OH5A9A1 - ASSY, KY58 PANEL
    - [ ] OH5A10A1 - ASSY, RIGHT CIRCUIT BREAKERS
    - [ ] OH5A10A2 - ASSY, CANOPY SWITCH
- [ ] Add items to Interconnect Drawings
    - [ ] OH5A1 - ASSY, RIGHT CONSOLE
    - [ ] OH5A1A1 - ASSY, RIGHT CONSOLE STRUCTURE
    - [ ] OH5A2A1 - ASSY, HOOK LEVER & INDICATOR
    - [ ] OH5A2A2 - ASSY, WING FOLD HANDLE
    - [ ] OH5A2A3 - ASSY, AV COOL SWITCH
    - [ ] OH5A2A4 - ASSY, RADAR ALTIMETER GAUGE
    - [ ] OH5A2A5 - ASSY, HYD PRESS GAUGE
    - [ ] OH5A3A1 - ASSY, CAUTION LIGHT PANEL
    - [ ] OH5A4A1 - ASSY, ELEC PANEL
    - [ ] OH5A5A1 - ASSY, ECS PANEL
    - [ ] OH5A5A2 - ASSY, DEFOG/WINDSHIELD PANEL
    - [ ] OH5A6A1 - ASSY, INTR LT PANEL
    - [ ] OH5A7A1 - ASSY, SNSR PANEL
    - [ ] OH5A8A1 - ASSY, SIM CNTL PANEL
    - [ ] OH5A9A1 - ASSY, KY58 PANEL
    - [ ] OH5A10A1 - ASSY, RIGHT CIRCUIT BREAKERS
    - [ ] OH5A10A2 - ASSY, CANOPY SWITCH
## CENTER TUB
- [ ] Mechanical Drawings
    - [ ] OHXXX - ASSY, CENTER TUB
- [ ] Interconnect Drawings
  - N/A (?)
## SEAT v1.5
- [ ] Mechanical Drawings
  - [ ] TBD
- [ ] Interconnect Drawings
  - [ ] TBD
## THROTTLE
- [ ] Mechanical Drawings
    - [ ] OH4A9A1 - ASSY, THROTTLE QUADRANT
    - [ ] OH4A9A2 - ASSY, OUTBOARD THROTTLE GRIP
    - [ ] OH4A9A3 - ASSY, INBOARD THROTTLE GRIP
- [ ] Add items to Interconnect Drawings
    - [ ] OH4A9A1 - ASSY, THROTTLE QUADRANT
    - [ ] OH4A9A2 - ASSY, OUTBOARD THROTTLE GRIP
    - [ ] OH4A9A3 - ASSY, INBOARD THROTTLE GRIP
## STICK
- [ ] Mechanical Drawings
  - [ ] TBD
- [ ] Interconnect Drawings
  - [ ] TBD
## TOP LEVEL ASSY
- [ ] Mechanical Drawings
  - [ ] TBD
- [ ] Interconnect Drawings
  - [ ] TBD

jrsteensen, 2021-08-03:
Removed interconnect drawings which are being bumped to beta2 milestone.

rmarrash, 2021-08-04:
Left and Right Console Packages
Issue: Correct angles on "aft Internal Bulkhead" required for smooth installation of remainder of ribs
1) Ensure that the angle of the cut on the bottom of the Aft Internal Bulkhead is 22.1 degrees. This will allow it to lie flat on the console floor at the correct angle and allow the installation of the rest of the ribs to go smoothly. 
2 ) Each Rib should be centered on holes in the console floor for perfect alignment.
[image]

rmarrash, 2021-08-04:
Left and Right Console
Issue: Correct angle of Canopy rail required for smooth transition/fit with Ribs and Rib Covering
Comment: The Angle needs to be 22.5 degrees. Also note, the width of the Canopy rail at its widest point after being cut needs to be ~90mm
[image]

[image]

rmarrash, 2021-08-04:
Lower Instrument Panel Construction
Issue: Angle at the back of "Upper Instrument Panel Mount" critical to proper installation of "right inboard footwell wall and left inboard footwell wall"
Comment: The angle of the cut on the back of the Upper Instrument Panel should be 13 degrees. This will allow the proper overhang of 3.175mm for the Right Inboard footwell wall and 12.78mm for the left Inboard Footwell wall. 
[image]
[image]
[image]

jrsteensen, 2022-09-21:
Going to track this per-issue in the beta 1 project.

closed 2022-09-21 by jrsteensen