jrsteensen/OpenHornet #318: Standby Instrument Panel monitor.

ddanois, 2021-07-28:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
(TFT, LCD, 11.6IN, 1920X1080P, COLOR monitor for Standby Instrument Panel may not be available. I placed an order with Aliexpress with a price listed at $179.00 two weeks ago. Today I received an email that my order had been canceled as the item was no longer available. When I checked the Aliexpress site, the monitor is still listed but it's now showing a price of $299. Not sure that this source is reliable and we may need another source or monitor)

### Expected Results/How It Should Work:
(Explain how you think it should work here. Be very specific.)

### Actual Results/How It Does Work:
(Explain how it actually works here. Be very specific.)

### Screenshots:
*Pictures are worth 1000 words. Take some screen grabs of the model, circle things with the snipping tool...you can't put too much info here.
<br>**PRO-TIP:** Just paste the images here.*

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [ X] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
*Check one item.*
- [ ] Bug
- [ X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

Ben-F111, 2021-07-28:
https://www.ebay.com.au/itm/202763964355

jrsteensen, 2021-07-29:
Displays to evaluate:

https://www.aliexpress.com/item/1005001658860376.html?algo_pvid=e7396022-36dc-404a-8eb7-2da9eba45061&algo_exp_id=e7396022-36dc-404a-8eb7-2da9eba45061-4

jrsteensen, 2021-09-02:
Will also evaluate a reversion to analog instruments.

jrsteensen, 2021-09-03:
After the vote on discord, it looks like we will be going the analog/digital route, with analog ALT/AS/VVI gauges, and a 7" LCD for the SARI/RWR. I will maintain the option to back the standby instrument module with an 11.6" screen in case people want to go with this option instead. #344 will track this effort.

closed 2021-09-03 by jrsteensen