jrsteensen/OpenHornet #319: Add WinWing Throttle Compatibility

jrsteensen, 2021-08-01:
TBD

L-Walker, 2021-09-26:
What this will involve:
1. Notch taken out of the rib next to it to stop you shaving your knuckles off on it as you go past MIL power.
2. Replacement inboard spacers forward and aft of the throttle.
3. Replacement outboard spacers between the ribs to allow the SIM PWR and EXT LT panel to slide outboard by 10mm to clear the port engine throttle lever.

L-Walker, 2021-10-09:
Having a winwing model would be worth its weight in gold. If anyone can help source one by either finding a pre-made one, hitting up winwing for one or making one.

jrsteensen, 2022-07-04:
This will be kicked forward to a future beta as mod packages. I forsee having mod packages for a number of winwing products due to their popularity.

closed 2026-09-20 by jrsteensen