jrsteensen/OpenHornet #321: Update/Add Selected L/R Console PCBs to Backlighting System

jrsteensen, 2021-08-01:
Update the following PCBs to use the Backlighting System:
**Left Console:**
- [ ] GEN TIE BACKLIGHT PCB
- [ ] UPDATE SIM PWR PANEL BL PCB

**Right Console:**
- [ ] GENERATE STEP/SLDPRT FOR CHECKLIST PLACARD AND ADD TO MCAD
- [ ] DESIGN AV COOL BL PCB AND ADD TO MCAD.

- [ ] ADD SLDPRT FROM ECAD TO MCAD FOR ALL BACKLIGHTING PCBS

## More Information

### Category:
- [x] ECAD (PCB Design or other electrical hardware)
### Type:
- [x] Feature Enhancement
### Applicable End Item:
*Check one item.*
- [X] Left Console
- [X] Right Console

### Associated Filename(s):
*Insert assembly or part file names here, i.e. 123456.sldasm, etc.*

jrsteensen, 2021-09-01:
@L-Walker Can you status this?

petersawka, 2021-09-10:
@L-Walker it appears the latest commit to Beta on the Right Console is missing the links to the following file:
[image]

L-Walker, 2021-09-12:
Yes you are quite right I need to move these to MCAD and out of the backlighting folder directory.

L-Walker, 2021-09-26:
Can someone correct me if I am wrong, but the file reference issues should be fixed now.

L-Walker, 2021-10-09:
SIM Power BL PCB has been created and is in the assembly. d57c7f3
AV Cool has been added to the right console and the relevant changes made. 03b97b8
Right console backlighting PCBs are all in MCAD. 66b039d

SIM PWR PCB

[image]
[image]

closed 2021-10-09 by L-Walker