jrsteensen/OpenHornet #322: Replace "toolbox" gears in ELEC & HYD PRESS.

jrsteensen, 2021-08-01:
*Please fill out the issue as completely as possible. Be very specific and take your time. The more effort you put in to fill the issue out completely, the quicker we can fix this or look at adding your requested feature.*

### Summary:
Replace the oversized gears with fixed gears in the ELEC & HYD PRESS Gauges.

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
### Type:
*Check one item.*
- [X] Bug
### Applicable End Item:
*Check one item.*
- [X] Right Console

L-Walker, 2021-10-09:
Talk to Ben about what he has done with this to make it work and potentially hoover up some part files for them.

L-Walker, 2021-10-09:
I am tapping out of this one, they need a bigger re-design than my skills can muster up.

jrsteensen, 2021-11-07:
Closed in referenced commit. (Eliminated gears by using dual needle microstepper.

commit 3d765f62 (2021-11-07): Completed ELEC/HYD PRESS Gauges

Closes #322, #257, and partially addresses #189.
closed 2021-11-07 by jrsteensen