jrsteensen/OpenHornet #324: Add cable clamps to ribs on consoles

jrsteensen, 2021-08-01:
### Summary:
Add Multiline Vibration-Damping Loop Clamps to ribs on R/L Consoles to route cable bundles into console from various components.

[Examples](https://www.mcmaster.com/cable-clamps/multiline-vibration-damping-loop-clamps/)

### Screenshots:
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)
### Type:
*Check one item.*
- [X] Feature Enhancement
### Applicable End Item:
- [X] Left Console
- [X] Right Console

L-Walker, 2021-09-12:
Using part: https://www.mcmaster.com/3225t24/

L-Walker, 2021-09-12:
Right console commit: #51866a

L-Walker, 2021-09-26:
Swapping the heat inserts for chipboard ones are on my to do list for tomorrow to complete this issue.