jrsteensen/OpenHornet #328: Backlighting Controller PCB Connector Arrangement

L-Walker, 2021-08-17:
The Molex mini-fit Jr connectors need to have at least 5mm more space between them in the Y axis to allow the connector clip to latch over the board header.

### Expected Results/How It Should Work:
Connector clips onto the board header

### Actual Results/How It Does Work:
Connector clip collides with the board header above it for rows 2 and 3 of the 2x2 connectors.

### Screenshots:
[image]

# Proposed Changes
[image]

### Suggested new header style to be used
[image]

## More Information
### Category:
- [ ] MCAD (SolidWorks)
- [ ] ECAD (PCB Design or other electrical hardware)
- [ ] Software - Sketch
- [ ] Software - DCS-BIOS
- [ ] Software - Library
### Type:
- [ ] Bug
- [ ] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
- [ ] Top Level Assembly          
- [ ] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
ABSIS Backlighting Controller?

jrsteensen, 2021-08-17:
@L-Walker, do you have a PN for the stackable headers (preferably LCSC/JLCPCB?)

L-Walker, 2021-08-17:
Good point, let me look up their catalogue and add it to our spreadsheet of JLC parts.

jrsteensen, 2021-08-20:
[image]

?

L-Walker, 2021-08-21:
Looks good to me

jrsteensen, 2021-08-28:
It was.

closed 2021-08-28 by jrsteensen