jrsteensen/OpenHornet #329: Cabin Pressure Mounting - suggestion

Ben-F111, 2021-08-21:
### Summary:
Alternate suggestion for the Cabin Pressure Alt

### Expected Results/How It Should Work:
A Minor changes is suggested to the Cabin Pressure Indicator Body to allow this to no longer require needing gluing at final assembly 

Extending the "Shadow" of the front Bezel to the Cabin Pressure Indicator Body (adding a 3mm lip) allowed this to be screwed 'Sandwiched' to the Lower Center Pedestal Face

see pictures below

### Actual Results/How It Does Work:
Currently it appears once the Cabin Pressure Gauge is constructed, there is no obvious way to attach the bezel to the main body other than gluing, this removes the future chances to correct any issues.

### Screenshots:
[image]
[image]

## More Information
*Add an "X" in the square brackets to check the applicable checkboxs.*
### Category:
*Check one or more items.*
- [X] MCAD (SolidWorks)

### Type:
*Check one item.*
- [ ] Bug
- [X] Feature Enhancement
- [ ] Maintenance
- [ ] Question
- [ ] Documentation
### Applicable End Item:
*Check one item.*
- [ ] Top Level Assembly          
- [X] Lower Instrument Panel (LIP)
- [ ] Main Instrument Panel (MIP)
- [ ] Left Console
- [ ] Right Console
- [ ] Seat
- [ ] Center Tub
- [ ] Flight Stick
- [ ] Throttle
- [ ] General Software
### Associated Filename(s):
Cabin Pressure Indicator Case

jrsteensen, 2021-08-21:
I like it. Lets do it!

jrsteensen, 2021-08-21:
Added a .125 mounting flange to the FWD edge of the case in F929cc9.

commit f929cc91 (2021-08-21): Cabin Pressure Gauge Mounting Flange

Added a .125 flange to the FWD edge of the Cabin Pressure Gauge Case.
Closes #329
closed 2021-08-21 by jrsteensen